guino/LSCOutdoor1080P #1: Is it possible for the LSC camera to be able to rotate?

Murat40, 2022-12-16:
Hey thanks for the great guide you made I really appreciate it. Now for me the question is if it is possible to make the camera Pan and Tilt. I tried this guide https://github.com/fjramirez1987/PTZ-YCC365 but I could not make it work. Is it because maybe there is no ONVIF in the camera or because I did something wrong?

Murat40, 2022-12-16:
I would love to help so if you can push me in the right direction when you have the time to do so. Maybe I could make the script of turning the motors. If it is possible to turn the motors from the Tuya app we should also be able to do so. I want it in my homeassistant setup but localtuya doesnt support it and I want it off the internet.

I will try to search for a solution and keep this Issue updated.

guino, 2022-12-16:
If you have coding experience I can give you some pointers as this will require a native C application to do make ioctl calls to the motor drivers (using the open handles in the main application) -- basically replicating the code that moves the motors in the tuya app.

Murat40, 2022-12-16:
I dont have too many coding experience, some from my bachelor in IT with python but I don't know if that helps any. Maybe the solution to the problem would be to search what code the camera executes when Tuya sends a datapacket to move the motors. This project actually seems pretty interesting and I don't know where to start.

guino, 2022-12-17:
@Murat40 I have just updated the repository files to include basic PTZ support for the rotating camera -- you should be able to use the provided URL to control the motion from Home assistant (assuming you can define the URL for each function).

Murat40, 2022-12-17:
@guino Thanks with my coding this would never be possible in a 100 years. Would you mind adding each other on discord? I would really want to learn the thing you can do but I don't know where to start.

guino, 2022-12-17:
@Murat40I barely use whatsapp and telegram, so I don't have discord. I try answering questions when people post here so you're welcome to do the same. The only time I use whatsapp/telegram is when I actually need to talk/solve something right then.

guino, 2022-12-17:
@Murat40 I forgot to mention: I added a change to motor.cgi so rotating the camera doesn't trigger motion detection -- just thought I'd mention in case that matters to you.

Murat40, 2022-12-17:
@guino I completely understand man! Also thanks for the change I will try tomorrow and give feedback if you want some ofcourse. Could you maybe give me a starting point then where I could start learning things like this? If it isn't much asked for.

guino, 2022-12-18:
@Murat40 if you want to mess with code you didn’t write, specifically embedded code you probably want to learn C coding paying special attention to pointers, memory allocation, function calling (by reference and by value), standard C libraries and data types. Then you can start playing with reverse engineering which is basically viewing the existing application into ‘poorly written’ C code with a tool like Ghidra - if you can manage to understand the the output from ghidra you can start trying to make minor changes which is the basis for what I have been doing in these projects. It can take quite some time but ghidra is a great tool to learn this stuff, specially if you write your own little applications for testing. A good ‘first goal’ would be to write a basic C application that asks for a password and verifies it’s a match or passes some sort of validation, then you can open the compiled application in ghidra and try to bypass the match/validation by modifying the compiled application (basically using ghidra output to modify the bytes in the file). This will take learning a little bit about machine language (assembly), at least enough to modify the code to do what you want. Not too long ago we had to so everything in assembly which was way harder, with ghidra you just need to learn a few assembly commands to control code flow and set parameters. From that you would have to mess with bigger and bigger code to get used to larger code layout and finding what you need.

alfakenzo, 2023-01-11:
Hi I have a question. It's maybe not that right topic. But for now it is possible to flash the WiFi chips beken etc.. with a custom firmware and to get Infos from a dump. Do you think there is a way with these to get more access in this cameras or doorbells from lac? For example openbeken or tuyacloucutter to flash the WiFi chip.
Maybe you know more and liem to answer.

guino, 2023-01-11:
@alfakenzo I'll be honest in saying that I had never heard of openbeken or tuyacloudcutter until you mentioned it above. I had heard of 'tuya convert' in the past but didn't really have much interest in it as I actually had manually flashed one of my tuya light switches (esp8266 based) in the past to custom firmware but that was just to play around with it.

To answer your question: The tuya **cameras** have very different structure than other more basic tuya devices (switches/lights/etc). I have only messed with the devices that have 'linux' firmware running (and there are a number of devices that seem to run RTOS firmware -- usually battery operated devices). From the linux devices I have seen, most of them seem to have RTL8188 wifi chips connected on a USB bus. So not only the hardware is very different but the wifi chip isn't the same either (and may vary from camera to camera as I have seen firmware with support for multiple wifi chips such as RTL8189 and ssv6x5x). At least on the cameras I have seen, I can tell you that the wifi chips are basically just slaves on the USB bus and they're only started late in the device's boot process, so I doubt that making any changes to them would offer any help in dumping the device's firmware or gaining access to the camera. 

There may be a way to exploit the tuya code handling data on ports 6668 and 6669 to trigger some code execution on the device allowing us to gain access to it -- I personally have not explored that option myself.

alfakenzo, 2023-01-11:
Ah ok but with the project Libretuya is also possible to flash, not OTA but hardwired, the RTL chips. RTL8710 and RTL8720 are in these project also possible to flash. So maybe you can take a look. But the app is using from thuya right? So could it be possible to write in the flash of the WiFi chip with 2mb all needed scripts.

guino, 2023-01-11:
@alfakenzo there's no '2mb flash' on the RTL8188 as far as I know it is just a dumb wifi-usb adapter:
https://www.realtek.com/en/products/communications-network-ics/item/rtl8188eus

The RTL8710 is a System-On-Chip hardware (with rom, ram, flash, i/o, etc):
https://www.realtek.com/en/products/communications-network-ics/item/rtl8710bn

The cameras have their own separate MCU (with flash, cpu, i/o, etc) and connect to a usb-wifi add-on (RTL8188).
The simpler devices use the wifi chip (RTL8710, etc) **AS** the MCU (with flash, cpu, i/o, etc).

To me this is basically trying to compare a full computer (RTL8710) to a USB-Dongle (RTL8188) that requires a computer to work.

The cameras use the tuya app (phone) and servers, yes -- but they communicate with a running application in the linux firmware (which is **not**running on the wifi chip), in fact there are a number of tuya cameras that have ethernet port which (when used) basically disable the wifi chip altogether.

Bottom line is I don't see anything useful in the **camera** wifi chips to gain access to the device. May be at some real low level someone could exploit some driver functions to cause code to execute in the OS but that would be a lot more work than messing with the camera flash directly.

alfakenzo, 2023-01-11:
Ah ok. This Infos are more clear for me. I will buy one of these cams and try your project.