guino/LSCOutdoor1080P #3: Access denied

tomtomtac, 2023-03-06:
Good morning,

I have a rotating LSC camera with version 2.10.28.
I applied option 3:
telnet works
rtsp works

but on the other hand impossible to access http://user:password@ip:8080
 access denied

I tried with a ppsFactoryTool is the problem is the same.

an idea?

Cdt

guino, 2023-03-06:
@tomtomtac For that URL to work you need to have busybox and httpd.conf in the SD card -- busybox should be downloaded from the link https://github.com/guino/LSC1080P/blob/main/mmc/busybox?raw=true and httpd.conf should have `/:user:password` (or you can edit user/password to what you want).

tomtomtac, 2023-03-07:
Yesss!
It works error on my part bad busybox!!
Thank you very much

closed 2023-07-31 by guino