guino/LSCOutdoor1080P #4: Version V3.10.57

Kocrucq, 2023-06-12:
Hello,

I have the same LSC outdoor1080P , i.e. the package is the same ! not rotatable camera
I have tried to begin with the hack  via option 1. but stopped because the md5 value did not match..
Can I use the 2.10.36 bin-file to go back to version V2.10.36   ?

Ko Crucq

guino, 2023-06-14:
@Kocrucq assuming you're running 3.10.57 (as in the issue subject) I would not try to 'downgrade' it to 2.10.36 as the differences may be too big for it to work.

If you stopped because the md5 was different I assume you got anyka_ipc file from your device ? if so, I'd say just go ahead and post a zip of it so I can take a look at it (hoping we can patch it).

If anything you can try option 3 from the repo (download/extract the 2.10.36 anyka_ipc and library + extract and run it WITHOUT modifying your flash). That should at least be safe to try without damaging the hardware/firmware but it will only work if the hardware/drivers are similar enough to 2.10.36 which there's no telling without seeing it.

Kocrucq, 2023-06-14:
Hello Guino,

 

Here is the anyko_ipc I found on the SD-card.

Many thanks in advance for your help.

 

Best regards Ko Crucq

michalfordonski, 2023-07-05:
Hi, I also have this camera with version 3.10.57.
As noted, I go through option 3 (paragraphs 1-6, binwalk, 7-16).

Unfortunately, it doesn't work for me - the camera lights up with a steady blue diode for 5 minutes and freezes without starting up. Reboot without SD unlocked the camera but RTSP is not working

I noticed that point 9 has a different checksum, even though the file according to the link from the manual was downloaded.
5941303f3338b4d0cf4332317ac2eb30

In addition, I would like to ask what needs to be set in the files from point 4?

Kocrucq, 2023-07-06:
Hi , your md5 value have to be the same, that is allmost by definition !!!! Did overwrite the anyka file ? as wagner suggest ? You did something wrong
Point 4. I only edit http.conf file for the wifi credential.

michalfordonski, 2023-07-06:
@Kocrucq I guess the manual doesn't quite describe the process correctly:

Option 3 dot 1 (Option 1 dots 1-6): create the original anyka_ipc file on the SD card
Option 3 points 2-3: downloading software from the server
Option 3 point 4: overwrite the file on the card with the software file from the link
Option 3 point 6 (option 1 points 7-16) Point 8: "select the original anyka_ipc file in the root of the SD" - but the file is already written on the card

So which file should you choose? the one created by the camera or the one from the link unpacked with binwalk?
The one from binwalk has a different checksum (5941303f3338b4d0cf4332317ac2eb30).

guino, 2023-07-06:
@michalfordonski you should be working with the file extracted from binwalk on steps 7-16. The md5sum should match otherwise it would mean the file wasn't extracted from binwalk. I have just now downloaded the bin file and checked and in fact the anyka_ipc extracted from binwalk has md5sum 5ac1f462bf039ec3c6c0a31d27ae652a 

Please note that binwalk will create several directories and the anyka_ipc extracted from the bin file will be under _165966791961ed11009a7.bin.extracted/_usr.sqsh4.extracted/squashfs-root/bin -- you have to copy just the anyka_ipc file from that location to the root of the SD card (you don't need any other file and it should not be inside any directories).

michalfordonski, 2023-07-06:
I formatted the SD card, reset the camera. I added it back to the app. I went through this process again this time with the correct checksum (previously the file was copied wrong and the checksum did not match). The camera has started. I can see it in the app. In the update tab, I have information about the possibility of updating to 3.10.57 - I had previously updated to this version and the message about the possible update no longer popped up. Unfortunately RTSP doesn't work - I tried rtsp://192.168.1.7/main_ch

Other links don't work either. NetScan only looks for port 6668

Do you have an idea what I can do to unblock RTSP?

michalfordonski, 2023-07-07:
Today I went through the whole process again. Same effect. Are you sure you should save the RomPatcherJS file with the note RTSP (anyka_ipc_rtsp) in step 11? Maybe I should save it as anyka_ipc and not anyka_ipc_rtsp?

I didn't change any files in step 4. The http.conf file does not exist. I'm guessing it's httpd.conf, but if I understand correctly, it's the login and password for the links: `http://user:password@ip:8080/cgi-bin/telneton.cgi` Right?

Kocrucq, 2023-07-07:
sorry, indeed it should be  httpd.conf. I used my user name and password for the LSC Smart connect app. Yes it should be anyka_ipc_rtsp. I followed exactly the procedure suggested by Wagner.

guino, 2023-07-10:
@michalfordonski I reviewed your files and everything seems. I then took your files after all steps, loaded them in my SD card and put it on my rotating camera and let it boot up. Everything worked as expected (rtsp, ptz control, file browsing, telnet access). So the problem is not with your files.

I will note it took a couple of minutes after booting for the device to be accessible on my network -- I waited for it to become online on the tuya app first, then I tried accessing the features.

I assume you have verified the device is working normally in the tuya/smartlife or lsc app before trying to use RTSP ? To be perfectly clear: I want to make sure you're not trying to use RTSP before setting up the device with the phone app as that's a requirement for the device to fully boot up. Once the device is setup in the app and you have RTSP working you have the option to never use the app anymore and even put if offline.

nailyk-fr, 2023-08-23:
Please find attached the `anyka_ipc` extracted by the script from version `3.10.57` :
[anyka_ipc.zip](https://github.com/guino/LSCOutdoor1080P/files/12418718/anyka_ipc.zip)
original md5sum : 
`e03dd3ae9c3ed20112371f2c3ff1f913  /media/nailyk/F745-FCC1/anyka_ipc`
(dunno if still useful or not)

nailyk-fr, 2023-08-23:
Here are my findings : 
Trying to patch the `anyka_ipc` extracted from `3.10.57` with https://github.com/guino/LSCOutdoor1080P/blob/main/anyka_ipc_rtsp.zip obviously doesn't work. 
Then I tried to use the `anyka_ipc` from `2.10.36` extracted with `binwalk` (+ the lib of course)
and everything is working: App + rtsp stream (`rtsp://ip:6668/main_ch`)

(Fun fact, the app is offering to update the firmware)

Next step: try to remove internet access as it is done in the other versions. I'll edit with the findings. 

Thanks sooo much for you amazing work on this !

winco206, 2023-11-09:
hello guys, Can anyonegive me all file for hack my camera?? I can not get binwalk to work. 
my camera has FW version 3.10.6x

guino, 2023-11-14:
@winco206 send me an email (my email is on my github profile) and I can try to help you out.

mr-brains, 2024-01-01:
> hello guys, Can anyonegive me all file for hack my camera?? I can not get binwalk to work. my camera has FW version 3.10.6x

I used 7 zip to ectract the files and that worked.