guino/LSCOutdoor1080P #5: Disable Tuya Cloud?

jeremypgn, 2023-07-14:
Hi! 

Thank you so much for your work! It allowed me to enable RTSP on a rotatable camera I bought for €30 at Action (a European store). Do you have a solution to stop sending the stream to the Tuya/LSC servers (to avoid being spied)? I want to use it exclusively locally.

Thanks in advance!

guino, 2023-07-15:
@jeremypgn I had not done much work on offline support for this device but I just did a test and it seems the device does boot up and start RTSP correctly even if you block the tuya servers, so you should be able to block the internet (or just tuya servers) on your router (which will prevent the phone app from working) and still use RTSP.

If you can't block the internet to the device on your router you can create a file on the SD card called `hosts` whit this in it (make sure the file has linux end of line markers):
```
127.0.0.1   localhost:localdomain   localhost
127.0.0.1 a.tuyaus.com a2.tuyaus.com a3.tuyaus.com m2.tuyaus.com mq.gw.tuyaus.com s.tuyaus.com
127.0.0.1 a.tuyacn.com a2.tuyacn.com a3.tuyacn.com m2.tuyacn.com mq.gw.tuyacn.com s.tuyacn.com
127.0.0.1 a.tuyaeu.com a2.tuyaeu.com a3.tuyaeu.com m2.tuyaeu.com mq.gw.tuyaeu.com s.tuyaeu.com
```
Then add this line to custom.sh (below the `touch /tmp/custom` line):
```
mount --bind /tmp/sd/hosts /etc/hosts
```

The above changes will block access to the tuya servers meaning the device will be off their cloud but still be accessible on your network.

Please notice:
1. A 'side effect' I noticed while doing the above: the status light will keep blinking indicating it is trying to connect to the tuya cloud (normally it stops blinking when it connects to the wifi). I suppose you can just put a piece of electrical tape over it if you don't want to see it blinking the whole time.
2. You will need to setup your own method for setting the time as the device uses the tuya cloud to do that. I added a command (below the `mount` command above), like this: `ntpd -p ca.pool.ntp.org` and it got the right time but in UTC time. Assuming you'll want to define the timezone I added the command (below the `ntpd` line): `echo EST5EDT > /tmp/TZ` and it adjusted the time to Eastern Time Zone. You will need to look at this list to find a setting for your timezone: https://docs.oracle.com/cd/E19057-01/nscp.cal.svr.35/816-5523-10/appf.htm.

This is what my custom.sh file looks like after all changes:
```
#!/bin/sh

# Mark that this runs
if [ ! -e /tmp/custom ]; then
 touch /tmp/custom
 mount --bind /tmp/sd/hosts /etc/hosts
 ntpd -p ca.pool.ntp.org
 echo EST5EDT > /tmp/TZ
 telnetd -p 24 -l /bin/sh
 /tmp/sd/busybox httpd -c /tmp/sd/httpd.conf -h /tmp/sd -p 8080
fi
if [ ! -e /tmp/cleanup`date +%Y%m%d` ]; then
 rm -rf /tmp/cleanup*
 touch /tmp/cleanup`date +%Y%m%d`
 /tmp/sd/cgi-bin/cleanup.cgi > /tmp/cleanup.log
fi
```

It should be clear that if you block the internet completely you'll need to create your own NTP server (locally accessible by the device) and adjust the ntpd command to point to your local server so you can set the time.

jeremypgn, 2023-07-16:
I hadn't thought about redirecting the traffic from the Tuya servers...

Thank you very much for all these details, I will hurry to test it!

closed 2023-07-16 by jeremypgn