guino/LSCOutdoor1080P #6: EMOS IP-300

jkolo, 2023-08-09:
I am requesting support to get RTSP up and running and get rid of cloud connectivity for an [EMOS IP-300 camera](https://www.emosgosmart.eu/en/products/gosmart-kamera-ip-300-torch/).

None of the ways I found on your github are working. The only thing is that I managed to [dump flash](https://github.com/guino/LSCOutdoor1080P/files/12307463/flash.bin.gz) using ppsMmcTool.txt (hopefully whole and correctly).

I also managed to connect to the camera [uart](https://github.com/guino/LSCOutdoor1080P/files/12307449/uart.log).

The [ppsFactoryTool.txt](https://github.com/guino/LSCOutdoor1080P/files/12307466/ppsFactoryTool.txt) file unfortunately does nothing. I am however also able to connect the camera to the network using the [ QRCode](https://github.com/guino/LSCOutdoor1080P/assets/1894746/a574db2f-0885-4a69-a1e3-dbbc3e1a71f9) and it is pingable. Unfortunately nmap shows that only port 6668 is open. I connected the camera to a wifi network without internet access. 

Below pictures of PCB:
[image]
[image]

guino, 2023-08-10:
@jkolo your firmware dump seems good. Do you know the exact firmware version you're running ? this looks like one of the newer firmware versions 5.2x or newer where they have removed the S80network script which allowed the boot settings to work.

You can probably patch your initrun.sh and flash it directly into the device to root it (assuming you have the tools/skills for it).

Your device should 'detect' ppsFactoryTool.txt from a quick look at it but please notice some features may only work if you have already added the device to the phone app (or the device will stay in a 'brand new' state which disables basically everything until you enroll it on the phone app).

guino, 2023-08-28:
@jkolo it is usually possible to flash using the bootloader, but it's always risky so using a hardware programmer allows you to make a backup and restore it if you do something wrong.

If you already have ONVIF, then likely you have RTSP too (it's basically part of ONVIF). Telnet/rooting will probably not help a lot, but you could poke around their code and make something to do motion detection (monitoring the log output) and control rotation (rotation control can take time though).

Most of the devices I've seen require you to remove the SPI flash from the board before reading/writing it, so keep that in mind as these boards are very fragile and you'll need the right tools and skills to do it (i.e. heat gun and some practice on old broken boards).