guino/LSCOutdoor1080P #11: Version 3.10.65 - Telnet working

busy-omega, 2024-02-04:
Just purchased a white LSC Smart Connect Outdoor IP camera in German Action Market (EUR 29.99)
The LSC Smart Connect app offers upgrade to version 3.10.67  (not yet upgraded).

The "210root.zip" method works and I get telnet access:
[root@anyka /usr/share]$ cat VERSION
3.10.65.1252
[root@anyka /usr/share]$ cat BUNDLE
HT_IPC202KS2M_TUYA_AK3918EV330

Followed steps 1 to 6 of Option 1 resulted in
anyka_ipc file with md5: a15d0ca41bc0694eda2af49ea97dfe05
attached as ZIP here

[anyka_ipc.zip](https://github.com/guino/LSCOutdoor1080P/files/14154649/anyka_ipc.zip)

Thats all for now. Any help appreciated to get digging further and hoping for rtsp support and cloud free / offline run.

guino, 2024-02-05:
I was just going to suggest using option 3 since we don’t have a patch specific for your version. Glad you got it working.

ricardowpv, 2024-05-17:
Just purchased the LSC Smart Outdoor Camera with external wifi antenna. I can confirm above steps also works for version 3.10.67.1
[image]

Version: 3.10.67.1
Bundle: HT_IPC202KS2M_TUYA_AK3918EV330

Marko9831, 2024-05-22:
can i have the file to downgrade?

guino, 2024-05-27:
@Marko9831 they're not downgrading it, they're just using option 3 I provided which runs an older version from the SD card (it doesn't downgrade the device). I sent you an email a few minutes ago, hopefully you can get option 3 working with the information on the email.

zethis, 2024-09-25:
Working on 3.10.73.1 too
`
[root@anyka ~]$ cd /usr/share/
[root@anyka /usr/share]$ cat VERSION
3.10.73.1
`