guino/LSCOutdoor1080P #12: Motion is not sending MQTT message ?

MaxCreeger32, 2024-03-20:
Hello, I am really a newbie in this type of hack. 
I have hacked my LSC Camera PTZ so easily with your tutorial. a big thanks for that ! 
I have modified the log_parser.sh to enter my MQTT topic and broker, but there is nothing on the topic.
I have activate the DEBUG_FILE and it has no effect and I do not understand why. 

here is the file : 

```shell
#!/bin/sh

DEBUG_FILE=/tmp/sd/output.log
DEBUG_FILE=$1

# contains(string, substring)
#
# Returns 0 if the specified string contains the specified substring,
# otherwise returns 1.
contains() {
    string="$1"
    substring="$2"
    if test "${string#*$substring}" != "$string"
    then
        return 0    # $substring is in $string
    else
        return 1    # $substring is not in $string
    fi
}

main() {
    IFS='$\n'
    echo -n "" > $DEBUG_FILE
    while true; do
        read -r BUF;
        if [ $? -ne 0 ]; then
            sleep 1;
            continue
        fi
        if contains "$BUF" " start new event"; then
            #echo "motion detected"
            /tmp/sd/mosquitto_pub -h 192.168.1.36 -u the_user -p the_password -m "1" -t homeassistant/LSCcamera/motion
        elif contains "$BUF" "##doorbell_push 3"; then
            #echo "doorbell push button"
            /tmp/sd/mosquitto_pub -h 127.0.0.1 -m "pushed" -t home/doorbell
        #else
            #echo "Unknown cmd: $BUF"
        fi
        echo $BUF >> $DEBUG_FILE
    done
}

main

```

thank you for your help if it is possible.

guino, 2024-03-29:
@MaxCreeger32 have you tried to execute the command `/tmp/sd/mosquitto_pub -h 192.168.1.36 -u the_user -p the_password -m "1" -t homeassistant/LSCcamera/motion` in telnet to see if it works ? that would be the first thing to check. The custom.sh file needs to be modified for log parser to work, you may want to check these links:
https://github.com/guino/BazzDoorbell/wiki/%5BHow-to%5D-Use-notifications-MQTT-log_parser-without-a-patched-ppsapp-%3F
https://github.com/guino/BazzDoorbell/wiki/%5BHow-to%5D-Integrate-with-Home-Assistant,-HomeBridge,-Domoticz,-etc

If you still have issues, please zip the contents of your SD card (without the SDT folder) and post it here so I can review it.

MaxCreeger32, 2024-05-13:
sorry for the late answer. yes the mqtt command works fine, I will check the links to have the Log parser working, thank you for all