guino/LSCOutdoor1080P #13: No RTSP on v2.10.36 from option 3

ThomasL-Dev, 2024-05-01:
I've recently buy one LSC Rotatable PTZ camera which was on version 3.10.75, i've followed the option 3 to set it on version 2.10.36, everything work well (i can control it from the "/ptz.html", telnet can be enable/disable from the url) except the RTSP flux, the camera still working on the mobile app, but can't get it from VLC, did i miss something from your readme ?

guino, 2024-05-07:
If you're running 2.10.36 and can control ptz, then you should be able to get RTSP from it as well -- VLC can be hit or miss with these devices depending on version/OS, etc. I would recommend you trying a different application, i.e. any phone app RTSP viewer, or ffmpeg/ffplay, or mplayer.

The url for playback should just be rtsp://192.168.1.nnn/main_ch (or whatever IP address for the camera), for instance:
`ffplay rtsp://192.168.1.83/main_ch`
`vlc rtsp://192.168.1.83/main_ch`

OrianaDavinci, 2024-09-08:
I also have the same problem. I used option 3 with 7-zip to recover the files. I can do telnet as well as control the camera via the web page.

on the other hand, RTSP does not work.

When I do a netstat -an I don't have any listening on port 554.