guino/LSCOutdoor1080P #14: Smart Solar IP Camera v4.0.117

BmdOnline, 2024-05-22:
I'm trying to customize [this model](https://www.action.com/fr-fr/p/3203968/camera-ip-solaire-lsc-smart-connect/).
[image]

Connecting with Tuya Smart, the camera is v4.0.117.
An upgrade to v4.0.120 is suggested. I don't install it.

What I've tried, without success :
- **option 3** from this repository
- method from **Merkury720/Merkury1080P** repository

Each time, only 6668 port is opened :
> $ nmap -p- 192.168.0.155
> Starting Nmap 7.94 ( https://nmap.org ) at 2024-05-20 00:26 CEST
> Nmap scan report for 192.168.0.155
> Host is up (0.041s latency).
> Not shown: 65534 closed tcp ports (conn-refused)
> PORT     STATE SERVICE
> 6668/tcp open  irc
> 
> Nmap done: 1 IP address (1 host up) scanned in 13.59 seconds

I'm using a Sandisk 64Gb, FAT32 MicroSD. Each time "DCIM" folder is created on the card, so it doesn't seem to be related to the card.

I don't know what else to do for the moment.
I will try to open the camera, but I don't know how to do without damaging it.

BmdOnline, 2024-05-22:
[image]
[image]

I've successfully opened the cam.
We have two sets of GND/TX/RX pins. I've checked they're not linked.
I will try both of them and see what happens.

BmdOnline, 2024-05-23:
I got results with the GND/RX/TX on the bottom left.

```
Ver:20220425-T31ZC

vol=>60
curr pir sens:2
curr dev volume:8 

spk_vol:45
func switch 0x4
[tuya] night mode:0   night_mode:0
low_power_value:10
[tuya] low_power_value 10 
reconnect times:0;
[1B][m[-][ INFO][ZRT_WIFI_Get_Config][428] get ssid:(removed) pwd:(removed)
[1B][m[1B][m[-][ INFO][cadc_open][41] adc open success.
[1B][madc:239
[-][ INFO][system][243] echo 59 > /sys/class/gpio/export, ret: 0, target: 127.0.0.1:58878
[-][ INFO][system][243] echo out > /sys/class/gpio/gpio59/direction, ret: 0, target: 127.0.0.1:48597
[-][ INFO][system][243] echo 0 > /sys/class/gpio/gpio59/value, ret: 0, target: 127.0.0.1:34384
auto
chn[0] 264
chn[1] 264
chn->0,r:1.000000, b=1152
chn->1,r:1.000000, b=480
jpg:0, 0
snap_delay 2
[1B][1;32;40mINFO(jzdl): jzdl version:1.4.0(00010400_4e61bfb)  built:20230907-1222(4.7.2 c)[1B][0m
[1B][m[-][ INFO][hal_init_slip][853] chn 0 set horizontal and vertical flip.
[1B][m[1B][1;32;40mINFO(persondet): Ingenic DL PersonDet Promotion Version:0.0.3(00000003_111fcc8)  built:20231102-2000(4.7.2 simd)[1B][0m
param frameWidth: 640 frameHeight: 360  sense: -1 score: 0.400000 detdist: 1 ptime: 1 count: 0 mod: 1 enable_perm: 0 enable_move: 0 track_mode: 0 observation_period: 0 active_count: 0 move_scale: 1.000000
warn: shm_init,53shm init already
[1B][m[-][ INFO][__uart_rev_process][920] battery:b 6a = 2922, fxg:04, rtc:19,t:9746
[1B][mAA hwinfo.battery=100 

cur adc 2922.000000 

[1B][m[-][ INFO][hal_ai_start][328] ai start success.
[1B][mwarn: shm_init,53shm init already
[__sensor_process] line:1841
rtc_cnt=5
[frame_pooling_thread--419 Channel:1 ]:1126(ms)
[IMP_Encoder_GetStream_Impl--2558 Channel:1 ]:1129(ms)
[1B][m[-][ INFO][hal_ao_start][288] ao start success.
[1B][m[1B][m[-][ INFO][mdevinfo_set_timezone][335] set timezone [8][CST-08:00]
[1B][maac_len=2
15 88 

Init aac encoder success :AAC max_out_buffer_len[768], input_pcm_frame_len[2048]
[1B][m[-][ INFO][__uart_rev_process][920] battery:b 5d = 2909, fxg:33, rtc:19,t:9747
[1B][mAA hwinfo.battery=100 

cur adc 2909.000000 

>power on<
audio_ready=1, cnt=0
[1B][m[-][ INFO][maudio_change_mode][217] change audio mode -1->0.
[1B][m[1B][m[-][ INFO][__uart_rev_process][920] battery:b 5e = 2910, fxg:30, rtc:7e,t:9747
[1B][mAA hwinfo.battery=100 

cur adc 2910.000000 

[1B][m[-][ INFO][_speak_process][74] start play audio:/system/voice/power_on(1).
[1B][m[1B][m[-][ INFO][hal_ao_set_volume][440] set volume 45
[1B][m[1B][m[-][ INFO][ZRT_WIFI_Get_Config][428] get ssid:(removed) pwd:(removed)
[1B][m[_led_process] line:377 pirï¼Œçº¢ç¯äº®1ç§’
[1B][m[-][ INFO][__uart_rev_process][920] battery:b 5e = 2910, fxg:30, rtc:7e,t:9747
[1B][mAA hwinfo.battery=100 

[... 89 lines cut ...]
  create=false
  update=true
  tag_path=/dev/mtdblock1
  cmdline=
  sensor_init=
  env=senv;[HW];init_vw=1920;init_vh=1080;nrvbs=2;mode=0;[SDK];fmode=0;[WIFI];SSID2=47616c617830;PASS2=575041304f5349524953;MAC=b8:ec:a3:dc:fa:2e;IP=192.168.0.155;CHANNEL=0;DNS1=192.168.0.1;IPSERVER=0.0.0.0;IPMASK=255.255.255.0;GATEWAY=192.168.0.1;LEASETIME=86400;dhcpc_ip_addr=192.168.0.155;dhcpc_ip_mask=255.255.255.0;dhcpc_gateway=192.168.0.1;dhcpc_dns_server=192.168.0.1;dhcpc_lease_time=86400;eenv;
  g_bootinfo=
  g_fwinfo=
```

MQTT ?
```
[1B][m[-][ INFO][_get_ip][18] local address 192.168.0.155:59521
[1B][m[1B][m[-][ INFO][_get_ip][24] server address 3.120.92.134:1883
[1B][m[1B][m[-][ INFO][mnet_scrab_start][331] svr ip:3.120.92.134 port:1883
[1B][mscrab start ok
```

key.cfg ?
```
[1B][0;32;31m[-][ERROR][muart_get_key][202] open key.cfg error!
```

I don't know how to access any bootloader or anything else. I just have logs.
I don't see anything referring to /mnt, sdcard, reading / writing.

mihovilkolaric, 2024-05-23:
Hi,

I am fighting with the same device, also without any success so far, also only port 6668:

```
Nmap scan report for 10.42.0.240
Host is up (0.045s latency).
Not shown: 999 closed ports
PORT     STATE SERVICE
6668/tcp open  irc
```

I performed the update to 4.0.120 (as I had issues on my WIFI on 4.0.117, and also could not convince the camera to record to the MicroSD-Card) - but the update did neither help with those issues, nor it changed anything in regards to the rooting-options I tried so far.

After the update, I found an update.bin on the Micro-SD:
[update.bin.zip](https://github.com/guino/LSCOutdoor1080P/files/15422318/update.bin.zip)
but could not find anything useful yet (using binwalk).

So far I tried:
- https://github.com/guino/LSCOutdoor1080P?tab=readme-ov-file#root-access
- https://github.com/guino/Merkury1080P?tab=readme-ov-file#conclusion

Unfortunately, I have no programmer, so I can't provide any flash-dump.

Any other hints how I could further analyze?

mihovilkolaric, 2024-05-24:
BTW: placing a file `_ak39_factory.ini` on the SD-Card does not lead to the effect described in https://github.com/guino/Merkury1080P/issues/42#issuecomment-1278036051 .

BmdOnline, 2024-05-24:
I've ordered a CH341A programmer, I will try to dump the firmware when I receive it.
Without desoldering the chip ([like this](https://www.alexbilz.com/post/2023-01-06-action-lsc-1080/)) if it works.

mihovilkolaric, 2024-05-25:
I might give it a try with an arduino - but I have no clue yet how to open the camera without damaging it.
Do you have any hints?

BmdOnline, 2024-05-25:
Remove the screw near of SD Card.
Then pull gently around the front of the camera.
Use a thin accessory that won't scratch. Possibly with your fingernails 😉 
The black side should start to emerge from the white frame.
Watch out for the cable used to connect the speaker.

guino, 2024-05-27:
@mihovilkolaric the update file you posted seems to have 'part' of the  root file system, but it doesn't seem to include the main application running on the device, settings or startup scripts. So there's not much we can do with it. If you do get a full flash dump I can take a look when I have a chance.

Getting into the bootloader should just be a matter of powering on (or power cycling) the device while you're connected to the RX/TX pins and pressing a key to stop the boot process and enter the bootloader. Many devices do have this disabled (it will boot no matter what you press), others have a password (it will ask for a password when you press something), and others just go right into the bootloader -- only way to know is trying.

BmdOnline, 2024-05-28:
When I press any key, I can interact but mixed with log output. It's difficult to deal with it.
I have to keep camera connected to tuya to prevent it to suspend.

```
Zeratul login: [1B][m[-][ INFO][set_wifi_status][96] WIFI_CONNECTED
[1B][m===value:169.[night<200, day>250]
[tuya_ipc_sd_get_status] line:147	curr sd_status:2 total:60875

count: 0
person detect time: 789ms

person_detect_switch = 1 
(...)

root
Password: [1B][m[tuya_ipc_sd_get_status] line:147	curr sd_status:2 total:60875

count: 0
person detect time: 790ms
(...)

dgiot010

count: 0
person detect time: 511ms

person_detect_switch = 1 

_person_detect_handle 740 w=640 h360 

(...)
Login incorrect
```

Then cleaned :
```
Zeratul login:
root (echoed)
Password:
dgiot010 (not echoed)
Login incorrect
```

I've tried passwords `dgiot010`, `telnet` and `root` without success.

I will receive CH341A programmer soon. I hope I can dump flash without desoldering.

guino, 2024-05-29:
I emailed you with information based on the log you provided, but just for future reference (to anyone else looking at this issue), the logs seem to indicate that this camera is **not** running Linux, likely it is running RTOS, meaning:a hardware programmer is the only option to make any modifications, and they would be very time consuming and limited to what's already compiled in the main application.

BmdOnline, 2024-05-30:
I have received my CH341A programmer and dumped entire flash : [XMC.zip](https://github.com/guino/LSCOutdoor1080P/files/15504624/XMC.zip)

mihovilkolaric, 2024-05-30:
@BmdOnline glad to read that you managed to get a flash-dump!
(I tried with `flashrom` on a raspberry pi, but as I have no SOP8-Clip (and bad soldering skills), the results were not stable/reproducible).

However, also with your dump, binwalk seems to recognize a lot of linux-related stuff, but - as for the `update.bin` posted earlier, it fails to extract a "full" root-filesystem.

```
binwalk -e XMC.bin 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
29048         0x7178          CRC32 polynomial table, little endian
30468         0x7704          LZO compressed data
31288         0x7A38          LZO compressed data
202000        0x31510         CRC32 polynomial table, little endian
206248        0x325A8         LZO compressed data
209836        0x333AC         Android bootimg, kernel size: 0 bytes, kernel addr: 0x70657250, ramdisk size: 543519329 bytes, ramdisk addr: 0x6E72656B, product name: "mem boot start"
622592        0x98000         uImage header, header size: 64 bytes, header CRC: 0x1A3A734D, created: 2023-11-22 06:28:42, image size: 5226788 bytes, Data Address: 0x80010000, Entry Point: 0x803CDE70, data CRC: 0xA7939E61, OS: Linux, CPU: MIPS, image type: OS Kernel Image, compression type: none, image name: "Linux-3.10.14-Archon"
2327946       0x23858A        bix header, header size: 64 bytes, header CRC: 0x628E0400, created: 2031-03-22 02:20:47, image size: 1048576 bytes, Data Address: 0x50A40000, Entry Point: 0x628E0400, data CRC: 0x73260000, image name: ""
3752610       0x3942A2        PGP RSA encrypted session key - keyid: 822CC2 401020 RSA (Encrypt or Sign) 1024b
4591776       0x4610A0        Linux kernel version 3.10.1
4657616       0x4711D0        DES SP2, little endian
4658128       0x4713D0        DES SP1, little endian
4676064       0x4759E0        LZO compressed data
4726848       0x482040        CRC32 polynomial table, little endian
5052572       0x4D189C        xz compressed data
5277248       0x508640        CRC32 polynomial table, little endian
5483462       0x53ABC6        PARity archive data - file number 21057
5844156       0x592CBC        ASCII cpio archive (SVR4 with no CRC), file name: "dev", file name length: "0x00000004", file size: "0x00000000"
5844272       0x592D30        ASCII cpio archive (SVR4 with no CRC), file name: "dev/console", file name length: "0x0000000C", file size: "0x00000000"
5844396       0x592DAC        ASCII cpio archive (SVR4 with no CRC), file name: "root", file name length: "0x00000005", file size: "0x00000000"
5844512       0x592E20        ASCII cpio archive (SVR4 with no CRC), file name: "TRAILER!!!", file name length: "0x0000000B", file size: "0x00000000"
7260836       0x6ECAA4        mcrypt 2.5 encrypted data, algorithm: "-=", keysize: 11527 bytes, mode: "=",
8825076       0x86A8F4        Unix path: /var/lib/jenkins/workspace/Release_IPC_SDK/src/tuya_ipc_sdk/ai_detect_storage(x
8852370       0x871392        Unix path: /var/lib/jenkins/workspace/Release_IPC_SDK/src/tuya_iot_sdk/wifi_cfg_serv/ez_mc.c
8877037       0x8773ED        AES Inverse S-Box
8888152       0x879F58        Certificate in DER format (x509 v3), header length: 4, sequence length: 1998
8901279       0x87D29F        AES S-Box
8907262       0x87E9FE        SHA256 hash constants, little endian
8932186       0x884B5A        AES Inverse S-Box
8932442       0x884C5A        AES S-Box
9115411       0x8B1713        Copyright string: "Copyright (C) 2J"
9171891       0x8BF3B3        ELF, 32-bit LSB processor-specific, ("")
9499250       0x90F272        Unix path: /sys/class/gpio/export
9539057       0x918DF1        mcrypt 2.5 encrypted data, algorithm: "^", keysize: 3934 bytes, mode: " ",
11069460      0xA8E814        mcrypt 2.5 encrypted data, algorithm: "", keysize: 26905 bytes, mode: "o",
12156928      0xB98000        uImage header, header size: 64 bytes, header CRC: 0xBA260EE2, created: 2021-08-07 09:15:35, image size: 2087483 bytes, Data Address: 0x80010000, Entry Point: 0x802EB500, data CRC: 0x247FD390, OS: Linux, CPU: MIPS, image type: OS Kernel Image, compression type: lzma, image name: "Linux-3.10.14-Immortal"
12156992      0xB98040        LZMA compressed data, properties: 0x5D, dictionary size: 67108864 bytes, uncompressed size: -1 bytes
14778368      0xE18000        Squashfs filesystem, little endian, version 4.0, compression:xz, size: 1155118 bytes, 52 inodes, blocksize: 131072 bytes, created: 2023-11-26 07:45:12
16252928      0xF80000        JFFS2 filesystem, little endian
```

@guino  with this new information given - do you still think it is RTOS, or are the chances that it is a linux?
The few extracted files at least look like linux ELF-binaries:

```file squashfs-root/bin/logcat 
squashfs-root/bin/logcat: ELF 32-bit LSB executable, MIPS, MIPS32 rel2 version 1 (SYSV), dynamically linked, interpreter /lib/ld-uClibc.so.0, stripped
```

BmdOnline, 2024-05-30:
@mihovilkolaric, I've removed my dump because it contains privacy data in it. 😞 
@guino, I can mail you access to the dump.

mihovilkolaric, 2024-05-30:
Good news!

[Unblob](https://unblob.org/installation/) succeeded in extracting the root-filesystem.
app_init.sh looks promising.

gtxaspec, 2024-05-31:
hi, if you guys need more info on this platform, you can join us here:
https://discord.gg/xDmqS944zr
we are experienced ingenic developers from the [thingino](https://github.com/themactep/thingino-firmware) project.  have fun!

mihovilkolaric, 2024-05-31:
Short update:
The camera executes a shellscript called other.sh when it is placed on the root-folder of the MicroSD-card.
With the path /tmp/mnt/sdcard/ you can access the SD-Card itself, and copy stuff from the "running filesystem" to the card.

E.g. the /etc/shadow looks like this:
```
root:dY6MT354.O2K.:0:0:root:/:/bin/sh
```

guino, 2024-05-31:
@mihovilkolaric did you try to start a telnet server from that other.sh script ? may need to put a mips busybox on the SD card to try and run telnetd from it.

mihovilkolaric, 2024-05-31:
@guino  yes, I was able to start a telnetd (at least according to `ps` and `netstat`). But until now I did all my tests with a factory-reset camera, so I could not try to connect.
This is the script I used:
[other.sh.txt](https://github.com/user-attachments/files/15517503/other.sh.txt)
And this is the output it produced:
[myhack.log.txt](https://github.com/user-attachments/files/15517504/myhack.log.txt)
(needed to rename both files to .txt as github neither allows .sh, nor .log)

Next step it to connect my cam to a WIFI, and see whether telnetd still starts.

BmdOnline, 2024-05-31:
> I asked @BmdOnline for a copy of the dump so I can take a look too.

I send you a link today. Have you received it ?

mihovilkolaric, 2024-05-31:
@guino , @BmdOnline :
Also after connecting to WIFI, the other.sh is executed.
So I replaced /etc/passwd and /etc/shadow with a version containing the hash from https://github.com/guino/LSCOutdoor1080P?tab=readme-ov-file#root-access and afterwards start `telnetd`.

The first two connection-attempts with user root password telnet did not work (camera closed the connection)
```
$ telnet 10.0.0.9
Trying 10.0.0.9...
Connected to 10.0.0.9.
Escape character is '^]'.

Zeratul login: root
Password: Connection closed by foreign host.
```
but then it worked:
```
$ telnet 10.0.0.9
Trying 10.0.0.9...
Connected to 10.0.0.9.
Escape character is '^]'.

Zeratul login: root
Password: 
Hello Zeratul!
[root@Zeratul:~]# 
```

So, the camera is now rooted!

BmdOnline, 2024-05-31:
Can you provide a working package ?

mihovilkolaric, 2024-05-31:
Sure.
Extract
[root_v1.zip](https://github.com/user-attachments/files/15518225/root_v1.zip)
to the SD-card, and boot with the card inserted.

mihovilkolaric, 2024-05-31:
Now that we have root-access - does anyone know how to enable motion-detection + recording to SD-Card?
(This is actually the reason why I started the investigation).

I guess it has to do with this files:
```
[root@Zeratul:tuya]# cat tuya_sd_record_on_off
1[root@Zeratul:tuya]# cat tuya_sd_record_mode
0[root@Zeratul:tuya]#
```
but changing them has no effect, and after a reboot there are reset to the default.

BmdOnline, 2024-05-31:
And how to enable rtsp stream...

BmdOnline, 2024-05-31:
@mihovilkolaric, I'm trying to reproduce your work, but I'm not familiar to `binwalk` and `unblob`
How are you extracting the root partition :
```
5844396       0x592DAC        ASCII cpio archive (SVR4 with no CRC), file name: "root", file name length: "0x00000005", file size: "0x00000000"
```

mihovilkolaric, 2024-05-31:
basically I just called
`$ unblob XMC.bin` which outputs
```
╭───────────────────────────────────────────────────────────────────────────────────────────────────── unblob (24.4.5) ─────────────────────────────────────────────────────────────────────────────────────────────────────╮
│ Extracted files: 75                                                                                                                                                                                                       │
│ Extracted directories: 38                                                                                                                                                                                                 │
│ Extracted links: 147                                                                                                                                                                                                      │
│ Extraction directory size: 24.17 MB                                                                                                                                                                                       │
│ Chunks identification ratio: 33.79%                                                                                                                                                                                       │
╰───────────────────────────────────────────────────────────────────────────────────────────────────────── Summary ─────────────────────────────────────────────────────────────────────────────────────────────────────────╯
            Chunks distribution             
┏━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━┳━━━━━━━━┓
┃ Chunk type          ┃   Size    ┃ Ratio  ┃
┡━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━╇━━━━━━━━┩
│ UNKNOWN             │ 15.69 MB  │ 66.21% │
│ CPIO_PORTABLE_ASCII │  2.27 MB  │ 9.58%  │
│ LZMA                │  1.99 MB  │ 8.40%  │
│ ELF32               │  1.84 MB  │ 7.77%  │
│ SQUASHFS_V4_LE      │  1.11 MB  │ 4.66%  │
│ JFFS2_NEW           │ 512.00 KB │ 2.11%  │
│ PADDING             │ 308.00 KB │ 1.27%  │
│ LZO                 │ 917.00 B  │ 0.00%  │
└─────────────────────┴───────────┴────────┘
               Encountered errors                
┏━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━┓
┃ Severity         ┃ Name                       ┃
┡━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━┩
│ Severity.WARNING │ ExtractCommandFailedReport │
└──────────────────┴────────────────────────────┘
```
and creates a directory `XMC.bin_extract/` and a lot of sub-folders, one of it being the root-filesystem.

BmdOnline, 2024-05-31:
Okay, I just have python issues. Will upgrade and try unblob.

BTW, root is working, but I have to keep camera connected (tuya app launched with camera preview) to prevent power sleep.

Thanks.

guino, 2024-06-01:
@BmdOnline I have never seen any of the 'linux' devices sleep before, I have heard of the battery operated devices having an automatic sleep to reduce power consumption.

Are you guys able to execute anything like `ls -laR /` or `ps -w` to identify the main application file ?

gtxaspec, 2024-06-01:
the platform you guys are working with is the Ingenic "Zeratul" battery platform, pretty common in the battery ip-cam world on the T series platforms...  the way it works, theres typically an external MCU controlling power to the SoC, and reading motion input from something like a PIR sensor.  When theres no motion, after a timeout the MCU will physically power down the SoC.  Theres 2 different processor architectures built in to the SoC.  RISC-V, and MIPS.  The RISC-V core will perform sensor and ISP initialization as soon as motion is detected, while the linux core boots up, and then takes over.

BmdOnline, 2024-06-01:
> the platform you guys are working with is the Ingenic "Zeratul" battery platform, pretty common in the battery ip-cam world on the T series platforms... the way it works, theres typically an [...]

Yes, it was something like this.
When motion is detected (pir sensor) or if tuya app is streaming video, camera stays online.
If not, it goes into deep sleep.
Linux restarts on each motion detection, reconnect wifi, and so on.

@gtxaspec, how [thingino-firmware](https://github.com/themactep/thingino-firmware) can help ?
@guino, I'm not sure you're receiving my emails ? Have you downloaded my flash dump ? I've mailed you a link yesterday.

gtxaspec, 2024-06-01:
So far we don't support the battery powered zeratul platform, it's complex to support since each different vendor can and do use different microcontrollers in their respective designs.  But we do have extensive documentation in case anyone wants to try and adapt and or create their own implementation.

mihovilkolaric, 2024-06-01:
@guino : bot `ps`
```
[root@Zeratul:~]# ps -w
  PID USER       VSZ STAT COMMAND
    1 root      1440 S    {linuxrc} init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW<  [kworker/0:0H]
    6 root         0 SW   [kworker/u2:0]
    7 root         0 SW   [rcu_preempt]
    8 root         0 SW   [rcu_bh]
    9 root         0 SW   [rcu_sched]
   10 root         0 SW   [watchdog/0]
   11 root         0 SW<  [khelper]
   12 root         0 SW<  [writeback]
   13 root         0 SW<  [bioset]
   14 root         0 SW<  [kblockd]
   15 root         0 SW   [irq/37-isp-m0]
   16 root         0 SW   [irq/38-isp-w02]
   17 root         0 SW   [kworker/u2:1]
   20 root         0 DW   [isp_fw_process]
   21 root         0 SW   [kworker/0:1]
   22 root         0 SW<  [cfg80211]
   23 root         0 SW   [kswapd0]
   24 root         0 SW   [fsnotify_mark]
   25 root         0 SW<  [crypto]
   39 root         0 SW<  [deferwq]
   40 root         0 SW   [kworker/0:2]
   86 root         0 SWN  [jffs2_gcd_mtd6]
   88 root         0 SW<  [kworker/0:1H]
   89 root         0 SW   [kworker/0:3]
   90 root      1732 S    /usr/bin/alps
   91 root      414m S    /stone/main
  118 root         0 SW   [kworker/u2:2]
  128 root         0 SW   [wl_event_handle]
  129 root         0 SW   [dhd_watchdog_th]
  130 root         0 SW   [dhd_dpc]
  131 root         0 SW   [dhd_rxf]
  152 root         0 SW   [wl_event_handle]
  187 root      1432 S    {mem_free.sh} /bin/sh /usr/bin/mem_free.sh
  237 root         0 SW   [mmcqd/0]
  339 root      1432 S    telnetd
  359 root      1440 S    /sbin/getty -L console 115200 vt100
 1082 root      1452 S    -sh
 2517 root      1424 S    sleep 3
 2533 root      1432 R    ps -w
```

and `netstat`
```
[root@Zeratul:~]# netstat -tulpn
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 0.0.0.0:6668            0.0.0.0:*               LISTEN      91/main
tcp        0      0 0.0.0.0:23              0.0.0.0:*               LISTEN      339/telnetd
netstat: /proc/net/tcp6: No such file or directory
udp        0      0 127.0.0.1:9097          0.0.0.0:*                           90/alps
udp        0      0 0.0.0.0:46767           0.0.0.0:*                           91/main
udp        0      0 10.0.0.9:54723          0.0.0.0:*                           91/main
udp        0      0 0.0.0.0:43486           0.0.0.0:*                           91/main
udp        0      0 127.0.0.1:41711         0.0.0.0:*                           91/main
udp        0      0 0.0.0.0:42750           0.0.0.0:*                           91/main
netstat: /proc/net/udp6: No such file or directory
```

show two interesting processes: `alps` and `main`.

You can find /usr/bin/alps in the mail I sent you yesterday - the interesting thing about "main" is, that it does not exist (on /stone/main) - the directory /stone/ is empty:
```
[... 63 lines cut ...]
187     /bin/busybox    /dev/console
187     /bin/busybox    /dev/console
187     /bin/busybox    /usr/bin/mem_free.sh
339     /bin/busybox    /dev/null
339     /bin/busybox    /dev/null
339     /bin/busybox    /dev/null
339     /bin/busybox    socket:[813]
339     /bin/busybox    socket:[885]
339     /bin/busybox    /dev/ptmx
359     /bin/busybox    /dev/console
359     /bin/busybox    /dev/console
359     /bin/busybox    /dev/console
1082    /bin/busybox    /dev/pts/0
1082    /bin/busybox    /dev/pts/0
1082    /bin/busybox    /dev/pts/0
1082    /bin/busybox    /dev/tty
8855    /bin/busybox    /dev/null
8855    /bin/busybox    /dev/console
8855    /bin/busybox    /dev/console
[root@Zeratul:~]#
```

I have no idea where this `/stone/main` comes from, resp. why it vanishes after starting ...

I can send you `ls -laR` tomorrow, when I have physical access to the SD-Card - but in yesterday's mail you can find a copy of most folders.

mihovilkolaric, 2024-06-01:
btw, `dmesg` reports:
```
[root@Zeratul:~]# dmesg 
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.10.14-Archon (biao@ubuntu) (gcc version 4.7.2 (Ingenic r2.3.3 2016.12) ) #17 PREEMPT Wed Nov 22 14:28:41 CST 2023
[    0.000000] CPU0 RESET ERROR PC:D2C3289D
[    0.000000] CPU0 revision is: 00d00100 (Ingenic Xburst)
[    0.000000] FPU revision is: 00b70000
[    0.000000] cgu_get_rate, parent = 1104000000, rate = 0, m = 0, n = 0, reg val = 0x081000ff
[    0.000000] cgu_get_rate, parent = 1104000000, rate = 0, m = 0, n = 0, reg val = 0x081000ff
[    0.000000] CCLK:1104MHz L2CLK:552Mhz H0CLK:200MHz H2CLK:200Mhz PCLK:100Mhz
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 004ce000 @ 00010000 (usable)
[    0.000000]  memory: 00032000 @ 004de000 (usable after init)
[    0.000000] User-defined physical RAM map:
[    0.000000]  memory: 02800000 @ 00000000 (usable)
[    0.000000] Initial ramdisk at: 0x80600000 (13278208 bytes)
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000-0x027fffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x00000000-0x027fffff]
[    0.000000] On node 0 totalpages: 10240
[    0.000000] free_area_init_node: node 0, pgdat 804dd4f0, node_mem_map 812ab000
[    0.000000]   Normal zone: 80 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 10240 pages, LIFO batch:1
[    0.000000] Primary instruction cache 32kB, 8-way, VIPT, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 8-way, VIPT, no aliases, linesize 32 bytes
[    0.000000] pls check processor_id[0x00d00100],sc_jz not support!
[    0.000000] MIPS secondary cache 128kB, 8-way, linesize 32 bytes.
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping off.  Total pages: 10160
[    0.000000] Kernel command line: console=ttyS0,115200n8 mem=40M@0x0 rmem=24M@0x2800000 root=/dev/ram0 rw rdinit=/linuxrc mtdparts=jz_sfc:256K(boot),352K(tag),5M(kernel),6M(rootfs),2560K(recovery),1440K(system),512K(config),16M@0(all) lpj=6955008 quiet senv;[HW];init_vw=1920;init_vh=1080;nrvbs=2;mode=0;[SDK];fmode=0;[WIFI];SSID2=**********;PASS2=****************;MAC=************;IP=10.0.0.9;CHANNEL=0;DNS1=10.0.0.138;IPSERVER=0.0.0.0;IPMASK=255.255.255.0;GATEWAY=10.0.0.138;LEASETIME=86400;dhcpc_ip_addr=10.0.0.9;dhcpc_ip_mask=255.255.255.0;dhcpc_gateway=10.0.0.138;dhcpc_dns_server=10.0.0.138;dhcpc_lease_time=86400;eenv; lzo_size=5917331 rd_start=0x80600000 rd_size=0xca9c00
[    0.000000] ir_switch_parse mode: 2 threshold min:2000 max:2500
[    0.000000] ir_switch_parse width:1920 height:1080 nrvbs:2
[    0.000000] ir_switch_parse hight framerate mode change num:5
[    0.000000] ir_switch_parse dayEv:0 nightEv:0 coeff:0 wbr:0 wbb:0
[    0.000000] Sensor Calibration Mode:0
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Memory: 21240k/40960k available (3867k kernel code, 19720k reserved, 1051k data, 200k init, 0k highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:358
[    0.000000] clockevents_config_and_register success.
[    0.000018] Calibrating delay loop (skipped) preset value.. 1391.00 BogoMIPS (lpj=6955008)
[    0.000030] pid_max: default: 32768 minimum: 301
[    0.000236] Mount-cache hash table entries: 512
[    0.000747] Initializing cgroup subsys debug
[    0.000772] Initializing cgroup subsys freezer
[    0.002756] NET: Registered protocol family 16
[    0.003172] register process exit failed.
[    0.012704] bio: create slab <bio-0> at 0
[    0.014404] jz-dma jz-dma: JZ SoC DMA initialized
[    0.014756] media: Linux media interface: v0.10
[    0.014810] Linux video capture interface: v2.00
[    0.014835] TTFF tx_isp_riscv_hf_prepare:240 tx_isp_stop_riscv start:400
[    0.014845] AE Para ADDR = a3700000 HEAD:3aca3aca SENSOR:3aca3aca
[    0.014859] Current sensor again:986331842 isp gain:986331850 inttime:986331850 luma:986329794 ev:2977
[    0.014867] Current awb rgain:986329794 bgain:986331850
[    0.014874] Riscv frame count:986331850
[    0.014879] Tuning mode is:1
[    0.014937] isp_memopt = 0
[    0.014977] isp_ch0_pre_dequeue_time = 0
[    0.015150] wait stable.[289][cgu_isp]
[    0.015504] isp_ch0_pre_dequeue_valid_lines = 900
[... 215 lines cut ...]
[    1.861893] jzmmc_v1.2 jzmmc_v1.2.0: err1 cmd=52 arg=00000c00 iflg=00002204 imask=fffffdfb status=1f000942
[    1.873041] jzmmc_v1.2 jzmmc_v1.2.0: err1 cmd=52 arg=00000c00 iflg=00002204 imask=fffffdfb status=1f000942
[    1.884174] jzmmc_v1.2 jzmmc_v1.2.0: err1 cmd=52 arg=00000c00 iflg=00002204 imask=fffffdfb status=1f000942
[    1.895344] jzmmc_v1.2 jzmmc_v1.2.0: err1 cmd=52 arg=00000c00 iflg=00002204 imask=fffffdfb status=1f000942
[    1.906487] jzmmc_v1.2 jzmmc_v1.2.0: err1 cmd=52 arg=00000c00 iflg=00002204 imask=fffffdfb status=1f000942
[    1.917648] jzmmc_v1.2 jzmmc_v1.2.0: err1 cmd=52 arg=00000c00 iflg=00002204 imask=fffffdfb status=1f000942
[... cut ...]
[    2.030129] jzmmc_v1.2 jzmmc_v1.2.0: err1 cmd=52 arg=80000c08 iflg=00002204 imask=fffffdfb status=1f000942
[    2.041291] jzmmc_v1.2 jzmmc_v1.2.0: err1 cmd=52 arg=80000c08 iflg=00002204 imask=fffffdfb status=1f000942
[    2.220068] mmc0: new high speed SDHC card at address e624
[    2.227411] mmcblk0: mmc0:e624 SU04G 3.69 GiB 
[    2.246430]  mmcblk0: p1
[    2.504828] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[root@Zeratul:~]#
```

(just anonymized SSID, passwd (they were in hex?) and MAC (plaintext))

mihovilkolaric, 2024-06-01:
@gtxaspec : that's interesting.
Do you have any idea how the app is able to wake up the linux-core? Do MCU and linux-core share the same WIFI?

gtxaspec, 2024-06-02:
in the designs we've seen, the microcontroller can be standalone, or built-in to the WiFi module, it depends on the design of the logic board. 

its not so much of an "app" waking up the linux core, but the design of the firmware built into the soc.

power on
both cores start up
risc-v starts up quick ( sub 300 ms )
risc-v initializes ISP while the linux core boots.

BmdOnline, 2024-06-02:
Other useful informations :
```
[root@Zeratul:bin]# uname -a
Linux Zeratul 3.10.14-Archon #17 PREEMPT Wed Nov 22 14:28:41 CST 2023 mips GNU/Linux
```

```
[root@Zeratul:~]# env
USER=root
TRANSFER_MODE=IIC
LD_LIBRARY_PATH=/system/lib:/usr/lib
OLDPWD=/usr
HOME=/
SENSOR=jxf37
CALLBACK_SCRIPT=
PS1=[\u@\h:\W]#
LOGNAME=root
TERM=vt102
PATH=/system/bin:/bin:/sbin:/usr/bin:/usr/sbin
SOC_TYPE=SOC_T31Z
SHELL=/bin/sh
PRODUCT_MODE=SINGLE
PWD=/
```

```
[root@Zeratul:~]# mount
rootfs on / type rootfs (rw)
proc on /proc type proc (rw,relatime)
devpts on /dev/pts type devpts (rw,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /dev/shm type tmpfs (rw,relatime,mode=777)
sysfs on /sys type sysfs (rw,relatime)
/dev/mtdblock6 on /config type jffs2 (rw,relatime)
/dev/mtdblock5 on /system type squashfs (ro,relatime)
/dev/mmcblk0p1 on /tmp/mnt/sdcard type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,usefree,errors=remount-ro)
```

```
[root@Zeratul:~]# df
Filesystem           1K-blocks      Used Available Use% Mounted on
tmpfs                    17200         8     17192   0% /dev/shm
/dev/mtdblock6             512       148       364  29% /config
/dev/mtdblock5            1152      1152         0 100% /system
/dev/mmcblk0p1        62336272     16816  62319456   0% /tmp/mnt/sdcard
```

```
[root@Zeratul:~]# fdisk -l

Disk /dev/mtdblock0: 0 MB, 262144 bytes
255 heads, 63 sectors/track, 0 cylinders
Units = cylinders of 16065 * 512 = 8225280 bytes

Disk /dev/mtdblock0 doesn't contain a valid partition table

Disk /dev/mtdblock1: 0 MB, 360448 bytes
255 heads, 63 sectors/track, 0 cylinders
Units = cylinders of 16065 * 512 = 8225280 bytes

Disk /dev/mtdblock1 doesn't contain a valid partition table

Disk /dev/mtdblock2: 5 MB, 5242880 bytes
255 heads, 63 sectors/track, 0 cylinders
Units = cylinders of 16065 * 512 = 8225280 bytes

Disk /dev/mtdblock2 doesn't contain a valid partition table

Disk /dev/mtdblock3: 6 MB, 6291456 bytes
255 heads, 63 sectors/track, 0 cylinders
Units = cylinders of 16065 * 512 = 8225280 bytes

Disk /dev/mtdblock3 doesn't contain a valid partition table

Disk /dev/mtdblock4: 2 MB, 2621440 bytes
255 heads, 63 sectors/track, 0 cylinders
Units = cylinders of 16065 * 512 = 8225280 bytes

Disk /dev/mtdblock4 doesn't contain a valid partition table

Disk /dev/mtdblock5: 1 MB, 1474560 bytes
255 heads, 63 sectors/track, 0 cylinders
Units = cylinders of 16065 * 512 = 8225280 bytes

Disk /dev/mtdblock5 doesn't contain a valid partition table

Disk /dev/mtdblock6: 0 MB, 524288 bytes
255 heads, 63 sectors/track, 0 cylinders
Units = cylinders of 16065 * 512 = 8225280 bytes

Disk /dev/mtdblock6 doesn't contain a valid partition table

Disk /dev/mtdblock7: 16 MB, 16777216 bytes
255 heads, 63 sectors/track, 2 cylinders
Units = cylinders of 16065 * 512 = 8225280 bytes

Disk /dev/mtdblock7 doesn't contain a valid partition table

Disk /dev/mmcblk0: 63.8 GB, 63864569856 bytes
255 heads, 63 sectors/track, 7764 cylinders
Units = cylinders of 16065 * 512 = 8225280 bytes

        Device Boot      Start         End      Blocks  Id System
/dev/mmcblk0p1               1        7765    62366720   c Win95 FAT32 (LBA)
```

unblob files :
`/etc/init.d/rcS`
```
#!/bin/sh

# Set mdev
echo /sbin/mdev > /proc/sys/kernel/hotplug
/sbin/mdev -s && echo "mdev is ok......"

# create console and null node for nfsroot
#mknod -m 600 /dev/console c 5 1
#mknod -m 666 /dev/null c 1 3

# Set Global Environment
export PATH=/bin:/sbin:/usr/bin:/usr/sbin
export PATH=/system/bin:$PATH
export LD_LIBRARY_PATH=/system/lib

# networking
ifconfig lo up

# Start telnet daemon
telnetd &

# Set the system time from the hardware clock
#hwclock -s

# Run init script
if [ -f /usr/bin/app_init.sh ]; then
    /usr/bin/app_init.sh &
fi

echo "Enter recover os"
```

`/usr/bin/app_init.sh`

```
#!/bin/sh

#reset_wifi
#sleep 1

insmod_wifi_2

sleep 6

for count in {1..10}
do
    if [ -b "/dev/mmcblk0p1" ] ; then
        break 
    fi

    if [ -b "/dev/mmcblk0" ] ; then
        mkdir -p /tmp/mnt/sdcard/
        mount /dev/mmcblk0 /tmp/mnt/sdcard/
        break
    fi

    sleep 1

done

ls /tmp/mnt/sdcard

if [ -f /tmp/mnt/sdcard/other.sh ]; then
    /tmp/mnt/sdcard/other.sh &
fi

#/usr/bin/firmup /dev/mtdblock7 1 1 1 1
#/usr/bin/firmup /emmc/update.bin 1 1 1 1
/usr/bin/firmup /tmp/mnt/sdcard/update.bin 1 1 1 1
sleep 1
reboot

#LED_GPIO_NUM=49
#led_blink.sh ${LED_GPIO_NUM} &

#tf_update.sh
#if [ $? -eq 1 ]; then
#    echo "Try AP update"
#    ap_update.sh
#fi

#sync

## reset wifi
#MODULE_DIR=$(uname -r)
#mkdir -p /tmp/modules/${MODULE_DIR}
#mkdir -p /lib/modules
#cd /lib/modules/
#ln -s /tmp/modules/*

#rmmod bcmdhd
#rmmod cywdhd
#reset_wifi

#reboot
```

mihovilkolaric, 2024-06-03:
@gtxaspec  My question was not precise enough - I am wondering how mobile app can wake up and connect to the camera, even if it is in sleep mode.

In a wireshark-trace I could see that, even though the cam does not respond to pings (and the telnet-session times out), it still is able to send and receive MQTT - and looks like this is used to wake it up.

mihovilkolaric, 2024-06-03:
@BmdOnline  the posted rcS is from the memory-dump - but when rooting and logging in, I see a different one:
```
[root@Zeratul:~]# cat /etc/init.d/rcS 
#!/bin/sh

#bootup_timer start >> /tmp/bootup_time
#ht bell open this
echo 43 > /sys/class/gpio/export
echo out > /sys/class/gpio/gpio43/direction
echo 0 > /sys/class/gpio/gpio43/value
echo 41 > /sys/class/gpio/export
echo out > /sys/class/gpio/gpio41/direction
echo 0 > /sys/class/gpio/gpio41/value
echo 62 > /sys/class/gpio/export
echo out > /sys/class/gpio/gpio62/direction
echo 0 > /sys/class/gpio/gpio62/value
echo 63 > /sys/class/gpio/export
echo out > /sys/class/gpio/gpio63/direction
echo 0 > /sys/class/gpio/gpio63/value

# Set Global Environment
export PATH=/system/bin:/bin:/sbin:/usr/bin:/usr/sbin
export LD_LIBRARY_PATH=/system/lib:/usr/lib

# Set transfer mode with MCU Environment (IIC or UART)
export TRANSFER_MODE=IIC

# Set Product mode with camera Environment (SUIT or SINGLE)
export PRODUCT_MODE=SINGLE

# Set callback script path Environment (default /usr/bin/shutdown.sh)
export CALLBACK_SCRIPT=

# Set Sensor Environment
export SENSOR=jxf37

# Set SOC Type Environment
export SOC_TYPE=SOC_T31ZC

bootup_timer prog_run >> /tmp/bootup_time

echo "" >> /etc/profile
echo "# Set transfer mode ENV" >> /etc/profile
echo "export TRANSFER_MODE=IIC" >> /etc/profile

echo "" >> /etc/profile
echo "# Set product mode ENV" >> /etc/profile
echo "export PRODUCT_MODE=SINGLE" >> /etc/profile

echo "" >> /etc/profile
echo "# Set callback script path Environment (default /usr/bin/shutdown.sh)" >> /etc/profile
echo "export CALLBACK_SCRIPT=" >> /etc/profile

echo "" >> /etc/profile
echo "# Set Sensor Environment" >> /etc/profile
echo "export SENSOR=jxf37" >> /etc/profile

echo "" >> /etc/profile
echo "# Set SOC Type Environment" >> /etc/profile
echo "export SOC_TYPE=SOC_T31Z" >> /etc/profile

hostname -F /etc/hostname

# Set mdev
echo /sbin/mdev > /proc/sys/kernel/hotplug
/sbin/mdev -s #&& echo "mdev is ok......"

# networking
ifconfig lo up &

# mount jffs2 rw partition
insmod_sfc
mount -t jffs2 /dev/mtdblock6 /config

# Format config patition if it is invalid
#if [ ! -f /config/.tag ]; then
#    echo "Format config partition..."
#    umount -f /config
#    flash_eraseall /dev/mtd6
#    mount -t jffs2 /dev/mtdblock6 /config
#    cd /config
#    cp -r /config_bak/* .
#    touch .tag
#    cd /
#    echo "Done"
#fi
mount -t squashfs /dev/mtdblock5 /system
if [ $? -ne 0 ]; then
    echo "mount system failed"
    #recovery
fi

if [ ! -d /config/profiles ]; then
        cp /config_bak/profiles /config/ -fra
        sync
fi

if [ ! -d /config/usb/ ]; then
        mkdir /config/usb/
        sync
fi

if [ ! -f /config/tuya/tuya_remove_on_off ]; then
        mkdir /config/tuya -p
        cp /config_bak/tuya/tuya_remove_on_off /config/tuya/tuya_remove_on_off
        sync
fi
#app
/usr/bin/alps &
/stone/main &

#wifi
insmod_wifi

#sd
sleep 0.01
insmod_vfat

echo 100 > /proc/sys/vm/swappiness
echo 16777216 > /sys/block/zram0/disksize
mkswap /dev/zram0
swapon /dev/zram0

#telnetd &
#cw2015
#insmod /lib/modules/pm_cw2015.ko
/usr/bin/mem_free.sh &
bootup_timer all_done >> /tmp/bootup_time

sleep 0.3
echo 1 > /sys/devices/platform/jzmmc_v1.2.0/detect_on
rm -f /lib/modules/cywdhd.ko
rm -f /lib/modules/3.10.14-Archon/cywdhd.ko
rm -f /lib/firmware/fw_bcm43438a1.bin
rm -f /stone/main &

sleep 6
if [ -f /tmp/mnt/sdcard/other.sh ]; then
    /tmp/mnt/sdcard/other.sh &
fi

sleep 0.5
/system/wl ampdu_mpdu 4
```

that one starts the applications `alps` and `main`, and afterwards deletes the "/stone"-folder.
Unfortunately, both happens before `other.sh` is executed, so I can't dump its content before removal.

So, I have still no idea where the "stone"-folder comes from - I did not ind it in any of the mtdblocks. Maybe there is some other memory/flash (or even ROM) in the SoC?

mihovilkolaric, 2024-06-03:
@guino 

> Are you guys able to execute anything like `ls -laR /` or `ps -w` to identify the main application file ?
here it is:
[ls-laR.txt](https://github.com/user-attachments/files/15540378/ls-laR.txt)

btw: The camera has no ftpd on it. Where could I find a busybox for this device including ftpd?

gtxaspec, 2024-06-04:
> @gtxaspec My question was not precise enough - I am wondering how mobile app can wake up and connect to the camera, even if it is in sleep mode.
>
> In a wireshark-trace I could see that, even though [...]

in the recommended hardware designs from ingenic, the wifi module can periodically check in at a predefined interval to the associated wifi A/P, while the linux portion is powered down.

gtxaspec, 2024-06-04:
> @guino
>
>
> here it is:
> [ls-laR.txt](https://github.com/user-attachments/files/15540378/ls-laR.txt)
>
> btw: The camera has no ftpd on it. Where could I find a busybox for this device including ftpd?

Try here https://github.com/gtxaspec/wz_mini_hacks/tree/master/SD_ROOT/wz_mini/bin

BmdOnline, 2024-06-04:
When we launch Tuya app (or LSC Smart Connect), the camera wakes up directly when connecting.
Maybe Tuya app send a specific network command, like some wake-on-lan ?
@mihovilkolaric, you don't see anything wiith Wireshark ?

gtxaspec, 2024-06-05:
> > Try here https://github.com/gtxaspec/wz_mini_hacks/tree/master/SD_ROOT/wz_mini/bin
> 
> Thx, that works!
> 
> Do you maybe also know how to prevent the linux from shutting down, and how to record the data from the camera?

the soc is physically powered down by external means, a standalone MCU or a combination WIFI/MCU.  which wifi module does your unit have? do you have up close photos of the board?

gtxaspec, 2024-06-05:
some basic documentation
https://github.com/themactep/wiki/tree/master/hardware/components/soc/ingenic/csdn/caibiao-lee

BmdOnline, 2024-06-06:
I have extracted all partitions from camera
```
dd if=/dev/mtdblock0 of=/tmp/mnt/sdcard/mtdblock0
and so on...
```

Reading [Partition Boot Analysis](https://github.com/themactep/wiki/blob/master/hardware/components/soc/ingenic/csdn/caibiao-lee/zeratul-1.md), and `tag` partition strings (see below) we have :

|partition | Size | Designation |
|-------|------|--------|
|mtdblock0 | 256 Ko | boot |
|mtdblock1 | 352 Ko | tag |
|mtdblock2 | 5 Mo | kernel |
|mtdblock3 | 6 Mo | rootfs |
|mtdblock4 | 2.5 Mo | recovery |
|mtdblock5 | 1.4 Mo | system (squashfs) |
|mtdblock6 | 512 Ko | config (jffs2) |
|mtdblock7 | 16 Mo | all |

`tag` partition contains some interesting parameters :
```
CMDLconsole=ttyS0,115200n8 mem=40M@0x0 rmem=24M@0x2800000 root=/dev/ram0 rw rdinit=/linuxrc mtdparts=jz_sfc:256K(boot),352K(tag),5M(kernel),6M(rootfs),2560K(recovery),1440K(system),512K(config),16M@0(all) lpj=6955008 quiet
FWIF"ZRT_release_202310261423_biao_ubuntu;[VERSION];ver=CAMERA_SOC_T31ZC_jxf37p_4.7.2_V;"
```
Camera soc is Ingenic T31ZC

Now we have to extract `rootfs` from `mtdblock3`.
`main` process may be stored here.

`The rootfs partition stores the root filesystem. rootfs is compressed with lzo and mounted through initrd. The initrd is similar to the initramfs method, and it is also a ramdisk, which is stored in the memory, so the files created or modified in the system only exist in the memory, and the content will be restored after the next startup.`

mihovilkolaric, 2024-06-06:
> Now we have to extract rootfs from mtdblock3.
> main process may be stored here.

Tried that (by copying  /dev/mtdblock to the SD-Card using telnet), but neither unblob, nor binwalk were able to extract anything out of the dumped file. I am out of ideas right now.

gtxaspec, 2024-06-06:
The header should be corrected, look at it with a hexeditor. 
https://github.com/gtxaspec/wz_mini_hacks/issues/69#issuecomment-1463610002

BmdOnline, 2024-06-07:
It works.

`rootfs` contains full `rcS` script and all removed files once running
```
/lib/modules/cywdhd.ko
/lib/modules/3.10.14-Archon/cywdhd.ko
/lib/firmware/fw_bcm43438a1.bin
/stone/main
```

Now we can access `main` process if needed.

BmdOnline, 2024-06-08:
I've tried to kill `main` process : camera restarts.
I've tried to copy `main` file into `stone` folder : camera restarts.

Maybe some files on sdcard may be created for enabling some features.
Or adding some parameters in config files... but I don't know what to look for.

Some stuff for investigating :
```
$ strings main | grep -F sdcard | sort | uniq
/mnt/sdcard/%s.%s.txt
/tmp/mnt/sdcard/
/tmp/mnt/sdcard/DCIM
/tmp/mnt/sdcard/DCIM/%04d/%02d/%02d
/tmp/mnt/sdcard/DCIM/%04d/%02d/%02d/
/tmp/mnt/sdcard/biz.dat
/tmp/mnt/sdcard/factory/
/tmp/mnt/sdcard/factory/factory.ini
/tmp/mnt/sdcard/factory/insmod_usb/insmod_usb.sh
/tmp/mnt/sdcard/factory/network_config.ini
/tmp/mnt/sdcard/factory_update
/tmp/mnt/sdcard/sn_play
/tmp/mnt/sdcard/tuya.dat
/tmp/mnt/sdcard/update.bin
insmod /tmp/mnt/sdcard/factory/asix.ko
insmod /tmp/mnt/sdcard/factory/dwc2.ko
insmod /tmp/mnt/sdcard/factory/usb-common.ko
insmod /tmp/mnt/sdcard/factory/usbcore.ko
insmod /tmp/mnt/sdcard/factory/usbnet.ko
rm /tmp/mnt/sdcard/tuya.dat
sdcard is full ,mkidr test fail
sdcard not exist!
sdcard_info
```

```
$ strings main | grep -F \.cfg | sort | uniq
/config/common//imp.cfg
/config/common/battery_last.cfg
/config/common/key.cfg
/config/common/net.cfg
/config/common/reconnect_info.cfg
/config/common/sys.cfg
/config/common/wifi.cfg
/config/dat/dev.cfg
/config/dat/factory.cfg
/config/dat/sn.cfg
open dev.cfg failed!
open factory.cfg error!
open key.cfg error!
open sn.cfg error!
open update.cfg error!
read file factory.cfg failed.
update.cfg
```

```
$ strings main | grep -F \.ini | sort | uniq
.init
/tmp/mnt/sdcard/factory/factory.ini
/tmp/mnt/sdcard/factory/network_config.ini
read factory.ini file failed.
```

```
$ strings main | grep -F \.dat | sort | uniq
!NULL==mask.data
.cpr.dat.hex
.data
.data.rel.ro
/conf/dat/biz.dat
/conf/dat/tuya.dat
/config/dat/biz.dat
/config/dat/ctcc.dat
/config/dat/dot.dat
/config/dat/tuya.dat
/store/biz.dat
/store/tuya.dat
/tmp/mnt/sdcard/biz.dat
/tmp/mnt/sdcard/tuya.dat
NULL==mask.data
b.data[i] != 0
biz.dat
calloc pcontainer->pelem[%d].data failed
cpr.dat
ctcc.dat
dot.dat
rm /tmp/mnt/sdcard/tuya.dat
tutk.dat
tuya.dat
```

```
$ strings main | grep -F https: | sort | uniq
# https://curl.haxx.se/docs/http-cookies.html
https://
https://%s%s/%s%s%s&v=1.0&bucket=%s&typath=%s
https://%s.%s%s?Expires=%d&OSSAccessKeyId=%s&Signature=%s&v=1.0&bucket=%s&security-token=%s
https://%s.%s%s?X-Amz-Security-Token=%s&X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=%s&X-Amz-Date=%s&X-Amz-Expires=60&X-Amz-SignedHeaders=host&bucket=%s&v=1.0&X-Amz-Signature=%s
https://a2-ueaz.tuyaus.com/d.json
https://a2-weaz.tuyaeu.com/d.json
https://a2.tuyacn.com/d.json
https://a2.tuyaeu.com/d.json
https://a2.tuyaus.com/d.json
```

```
$ strings main | grep -F .com | sort | uniq
*.tuya.com0
*.tuya.com0
*.tuya.com1
*.tuyaaf.com1
*.tuyaas.com1
*.tuyacn.com1
*.tuyaeu.com1
*.tuyain.com1
*.tuyajp.com1
*.tuyarf.com1
*.tuyasa.com1
*.tuyaus.com1
*.wgine.com1
.comment
ftp@example.com
https://a2-ueaz.tuyaus.com/d.json
https://a2-weaz.tuyaeu.com/d.json
https://a2.tuyacn.com/d.json
https://a2.tuyaeu.com/d.json
https://a2.tuyaus.com/d.json
iot_world@tuya.com1
m2.tuyacn.com
m2.tuyaeu.com
m2.tuyaus.com
mes.wisdomep.com
mq.gw.tuyacn.com
time.pool.aliyun.com
time.twc.weather.com
time.windows.com
```

```
$ strings main | grep -F .cn | sort | uniq
*.tuya-inc.cn1
cn.ntp.org.cn
de.ntp.org.cn
edu.ntp.org.cn
jp.ntp.org.cn
kr.ntp.org.cn
sgp.ntp.org.cn
tw.ntp.org.cn
us.ntp.org.cn
```

```
$ strings main | grep -F rtsp | sort | uniq
===============rtsp url:%s
rtsp
rtsp://%s:%d/h264/1/main/av_stream
rtspUrl
```

guino, 2024-06-10:
@BmdOnline when you killed the main process, did you try running it immediately after (from SD card) ? If we kill the main process on other cameras (ppsapp) the camera also restarts because of the watchdog, but (in most cases) if we start the main process again it keeps working -- which allows us to run a modified/patched version of the main application.

We even made a 'watchdog feeder' for the other camera types which allowed us to keep the device running while the main application was not running. I doubt the same code would work for this device, but I'm sure it would be possible to do something similar. I also have seen some separate tools (on specific devices) to disable the watchdog altogether, sometimes it's just a parameter in an executable with unrelated name (so only way to know is execute whatever tools may be available hoping to see some parameter information).

BmdOnline, 2024-06-10:
@guino, I've juste tried this script
```
set -x
kill -9 `pidof main`
kill -9 `pidof alps`
cp /tmp/mnt/sdcard/mtdblock3.rootfs/stone/main /stone/main
/usr/bin/alps &
/stone/main &
sleep 0.3
rm -f /stone/main &

```

Camera keeps alive longer, but restarts after approximately 15 seconds.
I've tried to play with `alps` and `main` scripts together, same behavior.

guino, 2024-06-14:
@BmdOnline I don't know if you've tried this yet but I assume you have tried it:
* without killing/re-running alps
* just running main from the SD card (without copying it first)
* copying main before killing it (to save time between killing/re-running it)

If you've tried the above and it none of it helped then we would have to modify the flash to run the main application from the SD card (when available) instead of running it from the flash memory -- otherwise there's no point in even trying to modify the main application.

mihovilkolaric, 2024-06-15:
Hi,

keeping the camera alive is rather easy ... 
looking at
`ls -l /proc/91/fd/
(91 is the pid of the main-process) showed that it accesses /dev/watchdog.

So I just tried
`watch -n 1 "echo 1 > /dev/watchdog"`
in one telnet-session, and in another I could kill main (connection to LSC-App failed of course), while the telnet-session kept alive for minutes:
```
root@Zeratul:~]# kill 91
[root@Zeratul:~]# date
Sat Jun 15 23:32:10 CST 2024
[root@Zeratul:~]# date
Sat Jun 15 23:36:02 CST 2024
[root@Zeratul:~]#
```

BmdOnline, 2024-06-16:
Yes, it works.
I imagine we can create a new camera software, or a patched version of `main` process, enabling `rtsp`, avoid use of tuya servers...

BTW, feeding watchdog seems to prevent camera to enter sleep mode.

gtxaspec, 2024-06-16:
You can try running thingino firmware on it.  The streamer and associated sources are fully open, as well as the build system being buildroot based.

mihovilkolaric, 2024-06-20:
Hi all,
I also did some tests in restarting the main-application.
Interesting finding: Copy `main` from sdcard before killing running instance leads to immediate reboot.

Thus I used following approach:
While feeding the watchdog by a shell-script rom another terminal, I used:
`[root@Zeratul:sdcard]# kill 91 ; cp stone/main /stone/ ; /stone/main > main_log.txt`
(I haven't touched `alps` so far)

Which lead to a reboot after ~15 sec, as @BmdOnline  described. However, it produced following log:
[main_log.txt](https://github.com/user-attachments/files/15910121/main_log.txt)

Some findings in the log:
Main complains about another missing file:
`[0;32;31m[-][ERROR][muart_get_key][202] open key.cfg error!`

(Looks like it is expected in /config/common/:
```
$ strings stone/main | grep key.cfg
/config/common/key.cfg
open key.cfg error!
```
but not present on my cam.)

and the last line that makes it to the logfile is:
`[06-19 22:26:25-- TUYA Err][mqtt_client.c:1283] __mq_recv err op_ret:-1 recv_len:0`

(no idea yet which o the two conditions trigger the reboot - but as I was still feeding it externally, it is rather an explicit reboot, than a watchdog-restart.

mihovilkolaric, 2024-06-20:
> You can try running thingino firmware on it. The streamer and associated sources are fully open, as well as the build system being buildroot based.

Do you know which existing model/build is similar to this hardware, and do you think that booting the "stock" linux, killing main, and running thingino's streamer (while feeding the watchdog externally) might work (as a "proof of concept")?

BmdOnline, 2024-06-20:
Binaries from [wz_mini_hacks](https://github.com/gtxaspec/wz_mini_hacks/tree/master/SD_ROOT/wz_mini/bin) are working.
For example, I can start `ffmpeg` from `wz_mini_hacks`.

@gtxaspec, maybe we can use `go2rtc_linux_mipsel` ?
How can we configure it without entire `SD_ROOT` package ?

Things to do :
- [ ] fully prevent camera restart
feeding watchdog prevent restat only for 5mn.
- [ ] wake up camera without tuya app
I've tried to analyse network request with `wireskark` but it's beyond my skills.
Maybe broadcast packet ? mac request ?
- [ ] Replace `main` process with another streamer
How to compile a standalone thingino's streamer ?
How to use `go2rtc` ?

gtxaspec, 2024-06-20:
1.if you keep feeding the watchdog and it reboots after 5 minutes, there could be an external mcu that has it's own watchdog.  can you please share photos of the entire pcb board, back and front?

3. go2rtc won't work without heavy modifications. you'll need a v4l video device to use ffmpeg to interface to go2rtc
or
feed go2rtc an rtsp stream.

you can download precompiled prudynt binaries from:
https://github.com/gtxaspec/prudynt-t/

BmdOnline, 2024-06-20:
[image]
[image]

[32 more replies]