guino/LSCOutdoor1080P #15: v3.10.68 issues

Marko9831, 2024-05-22:
on lsc outside 2.10.36 it has been working perfectly for 2 years!
Now I have bought a new one because I want to have another one.
But it just works with this one. this is version v3.10.68

if I delete the ''anyka_ipc_rtsp'' file the camera goes online.
If I add it it stays offline.

Anyone have any tips?

Marko9831, 2024-05-22:
I can't figure it out with Binwalk. I don't have Linux, and I have no idea how to do that with Windows

guino, 2024-05-27:
I sent you a reply to your email, check it and let me know.

elowine, 2024-05-31:
Hello, I just bought a camera.

It is in V3.10.68 but it is possible to upgrade to V3.10.68.

Does your script work with the above versions?

guino, 2024-05-31:
@elowine I believe someone reported [Option 3](https://github.com/guino/LSCOutdoor1080P#option-3----if-youre-not-on-21036) working with 3.10.6x, I don't have a device on that version to try but it should not hurt to try it.

elowine, 2024-06-01:
> @elowine I believe someone reported [Option 3](https://github.com/guino/LSCOutdoor1080P#option-3----if-youre-not-on-21036) working with 3.10.6x, I don't have a device on that version to try but it should not hurt to try it.

@guino  It works great, thanks a lot for the job.

However, I would like to configure the mqtt messages. I don't understand what to put in the file.

Do you have an example file?

guino, 2024-06-01:
@elowine The mmc files I provided for this device should already have the log_parser setup. All you have to do is adjust line 33 of log_parser.sh to have the required parameters for your mqtt server and client.

I recommend you connect to the camera using telnet and execute the mosquito_pub command from log_parser `/tmp/sd/mosquitto_pub -h 127.0.0.1 -m "detected" -t home/doorbell` adjusting the IP, message and type parameters to be what you need in whatever will be receiving the mqtt messages.

Once you have defined the commands the way you want, you just have to make sure line 33 has that same command in the log_parser.sh and it should notify when there are motion events.

The parameters are simple: IP is the IP of the mqtt server/broker you configured (can be any mqtt server/broker of your choice), the message and type can be whatever you want to use to identify the message from this specific device, you just have to configure these same events on the client (so it receives/parses them). 

How to configure the MQTT server/broker and client settings will largely depend on what software you're using (home assistant, home bridge, mosquitto, etc).