guino/LSCOutdoor1080P #16: V3.10.78 Not working!

X4V1T0, 2024-06-03:
Hi, I bought a LSC Rotatory Camera and it came with 3.10.78 version.

I've tried Option 3 but it doesn't do anything different, no RTSP, no Telnet.. nothing, it seems like is ignoring the SD card.

Did you managed to make it work with newer version than 3.10.36?

Thank you very much for your effort!!

guino, 2024-06-03:
@X4V1T0 I don't have a device on that firmware version to try it, but I have had reports that 3.10.6x has worked with option 3. Feel free to post a zip of your SD card contents so I can review it (or you can email it to me if you prefer - my email is on my github profile).

X4V1T0, 2024-06-03:
The problem is that when no anyka_ipc is created automatically, so I tried downloading the firm and using binwalk to extract as Option3 says. But in theory the file should be created by the camera, no? Thanks!

guino, 2024-06-04:
@X4V1T0 no, you need to extract anyka_ipc using binwalk for option 3.

WebWire-NL, 2024-08-17:
I have it working on 3.10.73.1
Are you sure you followed all the steps? Patch the old version from the link shared?
~ # cat /usr/share/VERSION
3.10.73.1
It's just very unstable trying to figure out why webcam stops at a certain moment.... i made a keep alive script to ping but doesn't seem to work.... camera just randomly stops.