guino/LSCOutdoor1080P #18: Not working on my side

ITMEDIAVISION, 2024-06-30:
I got the "sunpanel camera, Ol classic outdoor camera and the rotate camera all on latest version, all of them do not trigger to create the files mentioned in the steps, i also did a double check and it looks like ( correct me if im wrong ) the latest software got patched to trigger to .sh files or any other conf file, is a bummer only tthe doorbell supports the stream option ( for now )

guino, 2024-07-08:
I have not worked with any solar panel cameras before, most of those are battery operated which usually means they don’t run linux or run some sort of linux hybrid system that boots up when motion is detected or when required from the network. These devices are likely not a good option to stream RTSP continuously because that would drain the battery faster than it can charge.

I am always willing to work on new devices but I can’t do much if I don’t have it here.

WebWire-NL, 2024-08-16:
You can forget the solar one... it's MIPS based and has a Broadcom chipset (I opened it) I asked on discord and it's a very closed platform.