guino/LSCOutdoor1080P #19: V2.10.45

isotherm, 2024-07-23:
I have been working on this, trying to get root shell and possibly RTSP without permanently needing SD card. Please note that I have firmware V2.10.45, but looking at the squashfs, it seems very similar to V2.10.30. I think the below might still work.

Here's what I have so far:

1. Follow instructions at https://github.com/guino/Merkury1080P/issues/42#issuecomment-1317802613 to get a root shell.
2. Log in to the camera via telnet.
3. `vi /etc/config/passwd` and change the `x` between the colons to `20zVXVMtkhCxI`. (This password is `telnet`. To do something else, run `mkpasswd -m des` and use that output instead.)
4. `vi /etc/config/wifi_driver_new.sh` and add these two lines at the top of the `driver_install()` function, so it's like this:
   ```
   driver_install()                     
   {                                                                  
       tcpsvd 0 21 ftpd -w / -t 1800 &
       telnetd
       wifi_stop
   ```
5. Upon rebooting even without the SD card, telnet and FTP are available.

I'm working on RTSP next.

guino, 2024-07-23:
@isotherm this is cool. With RTSP the issue is that I had to write out a whole new function into the main application, and I only did that to the specific version I posted. 
In theory it should be possible to do it with other versions of the main application, but it is just easier/faster to use the version I already patched.

In theory it is also possible to write a script/tool that loads the main application to memory, patches it, then executes it (so you can fit everything in the config partition), but again that requires a lot of extra time.

You could potentially just download the patched main application from somewhere on the network, but either you need to use the ram (which is limited), flash (which doesn't have enough space) or some network mount (which may be limited by kernel), so using the SD card is the simplest/easiest way to do it.

I understand the benefit of not needing SD cards (believe me I prefer that as well), but I also see the benefit in being able to just remove the SD card and running the device in 'stock' mode.

isotherm, 2024-07-25:
Thank you for this explanation. While experimenting, I did have a case of the device no longer connecting to wireless. Without UART, I had to build an update.tar to reflash the /etc/config partition to a known state. If modifying anything in that script on the config partition, adding a failsafe to uninstall the hack (e.g. if a certain file exists on SD card) would be a good idea.

The following works to enable RTSP server via SD card without patching, and also avoid connecting to Tuya cloud at all:
1. On SD card, make empty file named `_ht_ap_mode.conf`.
2. On SD card, make file named `hostapd` with the following contents:
   ```
   #!/bin/sh
   killall udhcpd
   ifconfig wlan0 up
   /usr/bin/wpa_supplicant -Dwext -i wlan0 -c /mnt/wpa_supplicant.conf -B
   /sbin/udhcpc -i wlan0 -t 10 -A 5 -q &
   ```
3. On SD card, make file named `wpa_supplicant.conf` with the following contents (adjusted for local network):
   ```
   network={
        ssid="SSID"
        psk="PASSWORD"
   }
   ```
4. Reboot. Only open ports are 23 (telnet) and 554 (rtsp).

This could also be implemented, as above, to work without any SD card. The advantage is either not to need SD card at all, or not to worry about the hack being erased when the recorded video rolls over.

I'm still working on things like two way audio and the notifications. Also, I need to make sure wireless reconnects properly if the connection is lost.

Marco15453, 2024-11-17:
Is there any update here?

isotherm, 2024-11-17:
Straight one-way RTSP works when following the instructions given at [this comment](https://github.com/guino/LSCOutdoor1080P/issues/19#issuecomment-2250358892).

In theory it's possible to run a process on the camera to record by receiving its own RTSP feed. I tried this and it seemed to work ok, but I figured I would just record on a computer/server that's receiving the feed.

I have everything in IDA and know where the functions are that need modifying, so if @guino could share what modifications he needed to make and what new function had to be made, then I could make those changes to this binary also.

Marco15453, 2024-11-19:
Is it possible to still allow the camera to still connect to the cloud?

isotherm, 2024-11-19:
> Is it possible to still allow the camera to still connect to the cloud?

If you only want root, the instructions in the first post should work. If you will keep the SD card in the camera, you can skip steps 4 and 5, though I suppose the files might get erased when the card fills up and it starts deleting old videos/files.

guino, 2024-11-20:
> I have everything in IDA and know where the functions are that need modifying, so if @guino could share what modifications he needed to make and what new function had to be made, then I could make those changes to this binary also.

I wouldn't remember the changes myself by now. I do remember having to write a new function from scratch to get RTSP working -- the easiest way to find out what changed is to look at the patch file: https://github.com/guino/LSCOutdoor1080P/raw/refs/heads/main/anyka_ipc_rtsp.zip -- this just has the addresses and bytes changed from the original file. What I can suggest (and what I would be doing) is: 1-Download the original anyka_ipc file from Option 3, make a backup, apply the patch, then look look at the decompiled code (on the addresses in the patch), comparing with the decompiled original code (backup file).

I would give you 'source code' if I had anything but I wrote the changes (ARM assembly) in Ghidra command by command and tested until it worked, so the only thing I have is the manually patched anyka_ipc file -- which is the same exact thing you get when you apply the patch to the original anyka_ipc.

dynasource, 2025-04-24:
+1 would be nice to boot with RTSP without SDcards

isotherm, 2025-08-12:
> Hi there. [@guino](https://github.com/guino) [@isotherm](https://github.com/isotherm) thank you for your research. Did I understood properly, that I can make 1-4 steps and will get RTSP working? Is this a stable solution? LSC app works or I have chose LSC or RTSP ?

Yes, the steps in https://github.com/guino/LSCOutdoor1080P/issues/19#issuecomment-2250358892 work to run RTSP and it is stable. However, there is no audio. If you want to continue using LSC (or whatever) app, then you need to use a different method like the patch that is in the main repo. It leaves Tuya enable while also enabling the RTSP server.

I have a different method to get audio, by using the factory test, and I will try to document it here in the coming days.