guino/LSCOutdoor1080P #22: Rotable Camera (Dual Band version) also works

FlexxFR, 2024-09-01:
Very similar, same steps also works for the (newer?) dual-band version.
[image]

It shipped with 3.10.83 by default and there was an update available.
After the hack it runs on the patched 2.10.36.
RTSP works, PTZ controls also work.

NMAP scan shows now 2 telnet sessions are available:
port 23, busybox telnet
port 24 the anonymous telnet.

--> As the 2nd telnet has to be anoymous to save SD write actions (as explained somewhere), why is this not applicable for the busybox telnet?

There is also something running on port 8090. But not mentioned in this write-up. What is this?

FlexxFR, 2024-09-01:
Maybe I was to optimistic.
It works, but the config is not very stable. When the RTSP stream is accessed the cam crashed after a few minutes. I tried to connect it to frigate, it works, but after few minutes the feed crashed and the device is offline. A reboot works, but again only for a few minutes.

Is this a known issue? Is there anything to troubleshoot?

guino, 2024-09-05:
I think I have seen some people running 3.10.68 working without issues, but there are reports of instability on versions 3.10.78 and higher. We probably need to inspect the new application version and see if we can enable RTSP on it directly (without downgrading to 2.10.36), but if there's no RTSP code in it, there's not much that can be done. 

You should telnet in and check resources like memory usage, to make sure it it's not running out -- also, if you DONT connect to RTSP: does it continue to work normally (without crashing)?

FlexxFR, 2024-09-05:
Let's see. I disabled frigate, so RTSP is available (and working) but not used continuously. 
It still running the patched 2.10.36 firmware. App is working, normal HA plug-in is working. Let see, how stable this is.

This weekend I can re-enable RTSP and login with telnet and check resources and logs (if I can manage ;)). 

Would it be possible to get the low quality stream via RTSP (so it would use less resources?)

ricardowpv, 2024-09-07:
Yesterday I bought a new outdoor variant of this camera. This one is also dual band and shipped with 3.10.73. I used method 3 and had the same problems; it crashed after a few minutes with Frigate using the RTSP stream. 

I updated it to 3.202.8 by getting the SD-card out. After the update I put it back in and RTSP was still working. Its working for over 60 minutes now without crashing.
[image]

FlexxFR, 2024-09-07:
Interesting, but I don't understand it???
How could a firmware update, that you do not use when booting from ssd (because then still the older 2.10.36 patched version is used, have any effect?

At least, without RTSP used, the 2.10.36 firmware is stable and camera is still working via the app.

guino, 2024-09-07:
@ricardowpv Thank you for the information -- that's great to know, thank you for the feedback/information!

@FlexxFR I know this may be confusing but when we're doing option 3 we're not really using '2.10.36' entirely -- we're only using the 'main application' from 2.10.36 -- and the rest of the firmware (bootloader, OS, drivers, boot scripts, etc) are all from whatever firmware is installed on the device. So it very possible for the 2.10.36 main application to work better in some firmware versions and worse in other versions. If you are offered the same update version (3.202.8), you have basically nothing to lose in taking the SD card out, booting up, performing the upgrade, inserting the SD card and rebooting.

FlexxFR, 2024-09-07:
Ah, okay. Thanks for that explanation.

I did the upgrade, it's now running on 3.286.10, rebooting with SD card to enable RTSP again, re-enabled frigate and it appears to be better. At least it's running for a few hours straight enabled with frigate :) Let's see.

closed 2024-09-13 by FlexxFR