guino/LSCOutdoor1080P #24: Camera working fine, but only half an hour

rootje, 2024-09-17:
Hi there!

Great work on this project! I bought a bullet white LSC Camera, followed the steps for option 3 and the RTSP stream works great! But only for half an hour or so. After that the camera looses connection to the ip network. The only way to get it to work again is a powercycle.

How can I troubleshoot this issue? Is there logging that I can look into?

Please help.

rootje, 2024-09-18:
Seems that when I don't use the RTSP stream with my DVR, the camera stays online. As soon as I configure the RTSP stream in the DVR software the camera works for about half an hour.

guino, 2024-10-01:
There was a report of a similar issue by another user, please read here: https://github.com/guino/LSCOutdoor1080P/issues/22#event-14249988869

Bottom line was: there was an update available, it was installed and the device became stable after that, so if you have an update available, I would recommend you try it.

rootje, 2024-10-01:
I've returned it to the store already. It also suddenly decided to talk Chinese (at least I think it was Chinese) to me. Anyway, thank you for your reply. Maybe it helps someone else in the future!