guino/LSCOutdoor1080P #25: Can't get my 2.4/5 GHz LSCOutdoor1080P to rtsp, using Option 1- 3

milanov8, 2024-09-22:
The "210root.zip" method works and I get telnet access:
[root@anyka /usr/share]$ cat VERSION
3.202.9.22
[root@anyka /usr/share]$ cat BUNDLE
HT_IPC202KS2M_TUYA_AK3918EV330

Followed Option 1-3 resulted in no success
on Option 1: anyka_ipc file will not be written to SD card. only ".rc" file with zero bytes is new on SD card

any help appreciated

guino, 2024-10-01:
For 3.x firmware Option 1 will not work, you have to use Option 3.

Feel free to post a zip of your SD card contents (from Option 3) and I can review/check. I have not heard of a 3.202.9.22 version yet, so I'd ask you to check the version on the phone app (device update) - I have seem reports of 3.10.73 working.

milanov8, 2024-10-02:
On Option 3 the camera did not write anything to the SD Card.
I've already returned the camera at the action store and received the refund. 

I might get a new camera from the action shop to check the camera firmware with the Tuya app.
However, I'm pretty sure that also the Tuya app displayed an unusual version like 3.202.9.22

tbSRoe, 2025-01-05:
Thats right newest Version is: 3.202.9 in app :)

FanDjango, 2025-02-15:
Just FYI, bought such a camera yesterday.

```
[root@anyka /usr/share]$ cat VERSION
3.2023.92.1313
[root@anyka /usr/share]$ cat BUNDLE
HT_IPC202KSD3M_TUYA_AK3918EV330
```

**3.2023.92.1313**