guino/LSCOutdoor1080P #26: Chinese loop message

zethis, 2024-10-01:
My Cam is not booting up anymore after successful hack and 1hour of using it.

Name of the file played in loop: 8k16_cn_uuid_not_setup_uuid.mp3
Location of the MP3: ~/_165966791961ed11009a7.bin.extracted/_usr.sqsh4.extracted/squashfs-root/share

Same with or without SD card in it.

Reset process looks to work fine, but result by the same file played in a loop

Any solution to provide ?

guino, 2024-10-01:
The hack for the Outdoor and Rotating cameras doesn't modify the firmware in any way directly, so removing the SD card should allow it to boot normally (with unchanged OEM firmware). That said if you used option 3, it is possible for the different application version to modify the settings in a way that a newer version (from firmware) could not read it -- but I have never heard of a such issue with these devices. 

Do you remember what firmware version you have ? If it's a version others have used in the past there's a chance your device simply malfunctioned (hardware wise), and I highly doubt this is due to the hack but it could be due to strain on the hardware (continued RTSP streaming vs idle most of the time).

tty74it, 2025-02-07:
@zethis Have you resolved the chinese loop message. Today after 3 month of working camera i have the same issue