guino/LSCOutdoor1080P #28: LSC Smart Outdoor PTZ Camera V3.10.86

kecajtop, 2024-12-14:
I managed to gain access via telnet to this camera. I was able to copy file from and into SD card. Looks like that the hack.sh was partly executed.

kecajtop, 2024-12-14:
[root@anyka ~]$ df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 1.0M      1.0M         0 100% /
devtmpfs                 17.4M         0     17.4M   0% /dev
tmpfs                    17.4M      2.5M     15.0M  14% /tmp
tmpfs                    17.4M    160.0K     17.3M   1% /var
/dev/mtdblock7            4.6M      4.6M         0 100% /usr
/dev/mtdblock6          256.0K    236.0K     20.0K  92% /etc/config
/dev/mmcblk0             14.8G     40.0K     14.8G   0% /mnt

[root@anyka ~]$ cat /proc/cpuinfo
processor       : 0
model name      : ARM926EJ-S rev 5 (v5l)
BogoMIPS        : 351.43
Features        : swp half fastmult edsp java
CPU implementer : 0x41
CPU architecture: 5TEJ
CPU variant     : 0x0
CPU part        : 0x926
CPU revision    : 5

Hardware        : AK39EV330
Revision        : 0000
Serial          : 0000000000000000
[root@anyka ~]$

AlixerVolkov, 2025-01-07:
Amaizing!!!!!
following this topic

AlixerVolkov, 2025-02-03:
Dear @guino 
can you help us with this new LSC camera?
LSC Smart Outdoor Camera - 3206233
https://l66.pl/produkt/768282,kamera-obrotowa-lsc-smart-connect-do-uzytku-na-zewnatrz-1080p

AlixerVolkov, 2025-02-09:
I got rtsp with this file and it's working very well, but I don't know why is losting internet access (tuya app is not working and date cannot synchronize). It works perfectly with Home Assistant Local Tuya. Another think that I can see... 8090 port is published instead 8080 and hack web access is not working.

fyi @guino @kecajtop 

[anyka_ipc_v3_patched_test.zip](https://github.com/user-attachments/files/18724009/anyka_ipc_v3_patched_test.zip)

AlixerVolkov, 2025-02-09:
[image]

[image]

guino, 2025-02-26:
I am assuming the file you posted was from the Option 3 of the process I posted ? Chances are your device has different wifi hardware than the one that works with the Option 3 and it may be trying to get the status/reset/start the incorrect hardware. I would need the original anyka_ipc file from the camera (not from option 3) to see what's different, but I can't promise anything.

AlixerVolkov, 2025-02-27:
Dear @guino 
That's correct!!!!!