guino/LSCOutdoor1080P #31: New Action Outdoor IP Cam 1681PG 15  in White (SD Content)

tbSRoe, 2025-01-09:
Telnet is possible... 
Version is; 3.202.9

SD Card content is: 
[sd.zip](https://github.com/user-attachments/files/18363940/sd.zip)

Need RTSP :(-

Hope you can get it with this files👍

tbSRoe, 2025-01-09:
Keine ahung ob man das vllt braucht ?:

vi BUNDLE gibt aus:
HT_IPC202KS2M_TUYA_AK3918EV330

tbSRoe, 2025-01-10:
Joare methode 3 fuktioniert. Schlimmste war irgendwie binwalk zum laufen zu bekommen unter windows 11 xD 

Gäbe nur ne frage bezüglich MQTT bzw der motion detection... Da würde er mir irgendwie eine mqtt message senden wenn die "gute" erkennung von der kamera was erfasst korrekt ?

Stream geht audio höre ich auch, geht auch sprechen irgendwie umzusetzen ?

rkiendl, 2025-02-08:
habe die selbe Cam, bin aber ein wenig glücklos dazu. Hast Du die tuya anyka_ipc genommen oder die aus der SD Karte? Ich frage da die Größe der SD Karten  Version erheblich größer ist. tnx 4 info

tbSRoe, 2025-02-09:
Wie unter Option 3 beschrieben über binwalk extrahiert aus der Update Datei

tango38317, 2025-02-25:
Hi,
have same cam, but different firmware: v3.2023.92. tried also modding cam with option 3 but it seems, camera does not boot.
any idea how to get it up and running?
br
Chris

rkiendl, 2025-02-25:
were at the same spot but there ist hope: https://github.com/guino/LSCOutdoor1080P/issues/27
thanks to the gentle patcher there is also the binary in this issue threat and works great for actual outdoor cams :)

n1ghtlybuild, 2025-02-26:
Hi,
I just bought a [solar powered action camera from Action](https://www.action.com/de-de/p/3203968/lsc-smart-connect-solarbetriebene-ip-kamera/) and I was wondering if you might be talking about that model. However, I think "1681PG 15" is just part of the postal address of Action in Perenmarkt. My camera seems to have firmware version 1.4.50 and does not react to either 210root.zip content or the repository's mmc folder.

There are the following ports open on the device:
`PORT     STATE SERVICE
5002/tcp open  rfe
6668/tcp open  irc
8012/tcp open  unknown
8765/tcp open  ultraseek-http`

tango38317, 2025-02-28:
Hi, 
was not able to get it up and running with anyka_ipc_rtsp from #27  but its working with anyka_ipc_rtsp from #28 ?
does it work for someone of you with anyka_ipc_rtsp #27 ?
br
Chris

tango38317, 2025-02-28:
i have another issue with my camera, maybe some idea?
configured the camera and its working as expected, also with a lease from my dhcp range. after some days the camera has changed its ip address to a ip adress at the beginning of my subnet, but not in ip range? after rebooting, the camera has ip range again from dhcp range for some time and changes it again to another free ip outsid of my dhcp range?
some idea?
br
chris