guino/LSCOutdoor1080P #34: 2.4/5 GHz LSCOutdoor1080P

FanDjango, 2025-03-03:
I bought such a camera a few weeks ago and finally found time to work with it.

Option 3, Steps 1-7 **did not work, no response on telnet**. It did create the `hack` and `anyka_ipc` files on the mmc, though.

So I decided to try and find out more.

Without any mmc, using the LSC app, I can connect, see the video and hear the sound. The app reports the camera software version to be **3.2023.92**.

Using the appropriate discovery tools I can find the camera and its IP address in my network. In my case, I actually then caused my DHCP server to give the camera a fixed IP address by using the discovered MAC address.

Some research in the web allowed me to take the next tentative step:

I put together a small mmc (FAT formatted) with the following files in the root dir:

```
root@pluto:~/mmc# ls -l
total 4
-rwxr-xr-x 1 root root  0 Feb 15 13:33 _ht_av_tuning.conf
-rwxr-xr-x 1 root root 59 Nov 17  2022 shadow
```

`_ht_av_tuning.conf` is an empty file that causes `usr/sbin/service.sh`(in the camera's system) to "do something special", namely to activate the telnet daemon on **port 23**. The `shadow` file contains the encrypted user/password that `service.sh` then uses to authenticate logon to the system via the telnet daemon. The pre-coded user is ''root", the password is "telnet". 

Booting the camera with this mmc present, you can then login via telnet as root.

It then reports the following version information:

```
[root@anyka /usr/share]$ cat VERSION
3.2023.92.1313
[root@anyka /usr/share]$ cat BUNDLE
HT_IPC202KSD3M_TUYA_AK3918EV330
```

**3.2023.92.1313**

So far, so good. Now, Option 3, Steps 1-7 fails. Why? It uses a slightly different approach: It uses a different "special file" to tell `service.sh` to do something more useful (`_ht_ap_mode.conf` is the trigger in this case). This causes the camera to go into "access point mode" and to execute some files on the mmc. **This is much more useful than just activating a telnet daemon and it is the gateway to doing more**.

But why does this fail for this camera?

The camera has a different wifi driver and thus the file `hack.sh` on the mmc needs to be modified.

The main change is:

**Reverting from AP mode back to normal client mode**

Replace (in `hack.sh`):
```
# Reset wifi to client mode (same as it wifi_driver would have done)
killall udhcpd
ifconfig wlan0 0.0.0.0
touch /tmp/wifi_is_8188
```
by
```
# Reset wifi to client mode
/usr/sbin/ethernet.sh
```
Reason: `wifi_is_8188` is dependent on the hardware version and `/usr/sbin/ethernet.sh` is responsible for doing this right (see `/usr/sbin/service.sh` and especially `/usr/sbin/ethernet.sh`).

After this modification, the camera boots successfully with the mmc card present. It opens a telnet daemon and a busybox web server.

```
root@pluto:~# nmap 192.168.1.xx
Starting Nmap 7.93 ( https://nmap.org ) at 2025-03-03 12:11 CET
Nmap scan report for cam-007.xxx.yyy (192.168.1.xx)
Host is up (0.33s latency).
Not shown: 996 closed tcp ports (reset)
PORT     STATE SERVICE
23/tcp   open  telnet
24/tcp   open  priv-mail
6668/tcp open  irc
8080/tcp open  http-proxy
MAC Address: 38:7A:xx:xx:xx:xx (Unknown)

Nmap done: 1 IP address (1 host up) scanned in 3.47 seconds
```

You can logon via telnet on **port 24** (not on port 23!) without needing a user/password using `telnet 192.168.1.xx 24`.

You can access the html server via `http://192.168.1.xx:8080`. Use the user and password that you configured in `httpd.conf` (see step 4) to authenticate.

You can then use the busybox file listing to download the `anyka_ipc` file (or get this file from the mmc directly).

And that's it. Any attempt to go further does not succeed - the `anyka_ipc_rtsp` versions that you might try to use are not viable for this camera and thus the entire process of analysing and producing a patch still needs to be done.

I have attached the `anyka_ipc` file for this camera and this version (see above).

[anyka_ipc.zip](https://github.com/user-attachments/files/19052543/anyka_ipc.zip)

FanDjango, 2025-03-12:
Success!

I have made some more changes to `hack.sh` and the camera now streams RTSP, using the `anyka_ipc_rtsp` binary that you can find in the following place: [anyka_ipc_outdoor.zip](https://github.com/guino/LSCOutdoor1080P/issues/27#issuecomment-2509688395). You should perhaps check if @Ancientkingg has made any further changes or updates to this.

So actually a modified `anyka_ipc` from a different camera actually works - the code is probably quite generic, as already mentioned elsewhere.

The main problem that prevented the RTSP stream to work, was that `service.sh` continues to run alongside with `hack.sh` and it is a big matter of timing who finally starts the `anyka_ipc` (= wins the race, the original one or the `anyka_ip_rtsp` one that is bind-mounted in place of it).  

So in case the replacement loses out, a `pkill anyka_ipc && while pgrep -l anyka_ipc; do sleep 1;done;` will remove it, and the rtsp one will start.

I have also bought the indoor rotatable camera and it **also needed** this measure of killing the "wrong" anyka_ipc process.

The `netstat` **should** look like this, then:

```
~ # netstat -tlpen
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 0.0.0.0:23              0.0.0.0:*               LISTEN      443/telnetd
tcp        0      0 0.0.0.0:24              0.0.0.0:*               LISTEN      677/telnetd
tcp        0      0 0.0.0.0:8090            0.0.0.0:*               LISTEN      695/anyka_ipc_rtsp
tcp        0      0 0.0.0.0:554             0.0.0.0:*               LISTEN      695/anyka_ipc_rtsp
tcp        0      0 0.0.0.0:6668            0.0.0.0:*               LISTEN      695/anyka_ipc_rtsp
tcp        0      0 0.0.0.0:8080            0.0.0.0:*               LISTEN      680/busybox
```

Here's the complete `hack.sh` that I am using:

```bash
#!/bin/sh

# Mount SD on location that won't be umounted by device
mkdir -p /tmp/sd
mount /dev/mmcblk0p1 /tmp/sd

# Mark that this has run to this point
rm -f /tmp/sd/hack
if [ ! -e /tmp/sd/hack ]; then
 touch /tmp/sd/hack
fi

# If we don't have a copy of the application yet
if [ ! -e /tmp/sd/anyka_ipc ]; then
 cp /usr/bin/anyka_ipc /tmp/sd
 cp /usr/sbin/service.sh /tmp/sd
 cp /usr/sbin/ethernet.sh /tmp/sd
fi

# Remove AP mode file
rm -f /tmp/_ht_ap_mode.conf

# Reset wifi to client mode
killall udhcpd
ifconfig wlan0 0.0.0.0
/usr/sbin/ethernet.sh

# Mark that this has run to this point
rm -f /tmp/sd/ethernet
if [ ! -e /tmp/sd/ethernet ]; then
 touch /tmp/sd/ethernet
fi

# If we have a patched application, use it
if [ -e /tmp/sd/anyka_ipc_rtsp ]; then
 echo "export LD_LIBRARY_PATH=/lib:/usr/lib:/tmp/sd" > /tmp/anyka_ipc_rtsp
 echo "echo 'tilt_total_steps = 2200' >> /tmp/_ht_hw_settings.ini" >> /tmp/anyka_ipc_rtsp
 echo "/tmp/sd/anyka_ipc_rtsp 2>&1 | /tmp/sd/log_parser.sh /dev/null" >> /tmp/anyka_ipc_rtsp
 chmod +x /tmp/anyka_ipc_rtsp
 cp /usr/local/_ht_hw_settings.ini /tmp/
 mount --bind /tmp/_ht_hw_settings.ini /etc/config/_ht_hw_settings.ini
 mount --bind /tmp/anyka_ipc_rtsp /usr/bin/anyka_ipc
fi

pkill anyka_ipc && while pgrep -l anyka_ipc; do sleep 1;done;

# Run custom.sh on SD card every 10 seconds (like we do on other devices)
(while true; do /tmp/sd/custom.sh; sleep 10; done ) < /dev/null >& /dev/null &
```

Palcko, 2025-03-17:
Hello, I followed your instructions, but I'm unable to get a stream. The camera is working with the application.

FanDjango, 2025-03-17:
@Palcko 
What files are **now** on the SD card?
Can you telnet into the camera on port 24? And then issue command `ps -ef`, which `anyka....ipc` is running? What happens if you kill that PID number, after waiting a short moment do a `netstat -tlpen`, what is then the output?

FanDjango, 2025-03-17:
Ooops. Closed by mistake....

Palcko, 2025-03-17:
No, I can't connect. Here are the files on the card.

[Carte SD.zip](https://github.com/user-attachments/files/19295746/Carte.SD.zip)

FanDjango, 2025-03-17:
If you look at this code in the `hack.sh`:

```
# Mark that this has run to this point
rm -f /tmp/sd/hack
if [ ! -e /tmp/sd/hack ]; then
 touch /tmp/sd/hack
fi

# If we don't have a copy of the application yet
if [ ! -e /tmp/sd/anyka_ipc ]; then
 cp /usr/bin/anyka_ipc /tmp/sd
 cp /usr/sbin/service.sh /tmp/sd
 cp /usr/sbin/ethernet.sh /tmp/sd
fi
```
you will see that some files are written to the SD card (to prove that `hack.sh`) has run at all.

I don't see a file "hack" having been created on the SD card, and none of the others either.

So the SD card is not being reacted to.

It is interesting that ".rw" and "DCIM" have been created - this is done be the normal application. So, WRITING to the SD card seems possible, that can't be the problem.

One possible problem would be that the file "hack.sh" as been copied by you to the SD card but it does not have owner or execution privileges. I copied using a linux machine to the SD card (mounted it) and checked the files per "ls -l" command.

vratnik, 2025-04-13:
Sorry @FanDjango but your instructions are messy and unclear. You didn't clearly tell which steps you used. You should write up full own manual of steps you did and not refer to parts of not working other manuals.

FanDjango, 2025-04-13:
@vratnik 

It would be more constructive if **you** put in the effort to indicate what parts **you** _do not understand, or which parts do not work **for you**. Then one could work together to find out what is going wrong in **your** case.

>  You should write up full own manual of steps you did and not refer to parts of not working other manuals.

Huh? That's your opinion. In my opinion, there is no need to repeat a lot of stuff already explained and documented elsewhere. This is also work in progress with a moving target so you could be documenting this in an ongoing fashion, that's really not worth it right now.

closed 2025-04-24 by FanDjango

vratnik, 2025-04-26:
For example, you need to write it up in one place, without refering to various steps on various places. Therefore your solution didn't work for anyone. Btw you have an error in hack.sh line 21. But in my manal we won't need it:

1. Download the content of the repo from: https://github.com/guino/LSCOutdoor1080P/archive/refs/heads/main.zip
2. Extract the content from the “mmc” subfolder to the root of the microSD card;
3. Copy the Busybox file https://github.com/guino/LSC1080P/blob/main/mmc/busybox?raw=true and add it to the root of the microSD card
4. Put the microSD card in the camera, turn it on, wait 2-3 minutes (until the camera appears in the Tuya or LSC application), turn it off and remove the microSD card from the camera
5a. If you have the first version of the camera (2.4 GHz only), add the file anyka-ipc-rtsp [coming from here](https://drive.google.com/file/d/1FBRxn-UdJtuwufPp414tZpQqkBL1LlYI/view?usp=sharing) to the root of the microSD card.
5b. WARNING A new version of the camera has been on sale at Action since late 2024. This version includes the inscription "Dual Band" on the box and on the camera. For this model, it is necessary to use the file [whateverka-ipc-v3-patched-test.zip](https://github.com/user-attachments/files/17738739/anyka_ipc_v3_patched_test.zip) ([coming from here](https://github.com/guino/LSCOutdoor1080P/issues/27#issuecomment-2474702173)), rename it in anyka_ipc_rtsp
6. for PTZ camera, replace the file motor.cgi found here ([motor.zip](https://github.com/user-attachments/files/17739351/motor.zip)) in the folder cgi-bin on the SD card.
7. Safely eject the SD card and put it back in the camera
8. Start the camera and enjoy the stream at rtsp://\<camera ip\>/main_ch