guino/LSCOutdoor1080P #35: Issue with Rotatable Camera V3.2863.94

Fmaulick, 2025-03-29:
Hi;
I did each step
Option 1: 1-6
Option 3: 1-6
Option 1: 7-16
But my camera is not reacting after inserting the sd card.
Can someone help me?
Any Tips?

FanDjango, 2025-03-29:
Read #34. I have had success with the rotatable camera also. `hack.sh` needs some enhancments.

FanDjango, 2025-03-29:
I now have both these cameras and needed to modify `hack.sh` as described.

FanDjango, 2025-03-29:
https://github.com/guino/LSCOutdoor1080P/issues/34#issuecomment-2717996915

Fmaulick, 2025-03-29:
> I now have both these cameras and needed to modify `hack.sh` as described.

So you took, the anyka_ipc_outdoor.zip file.
And modified your hack.sh

So if I use your hack.sh and use the Zip file for the Option 1 step 7 (maybe 8, the one with the website)

The cam should work?

FanDjango, 2025-03-30:
@Fmaulick Yes

> So you took, the anyka_ipc_outdoor.zip file.
And modified your hack.sh

If it does not work, inspect the SD card to see which files have been produced, that will tell you that `hack.sh` has been executed or not.

Fmaulick, 2025-03-30:
I have an empty hack and ethernet file. But the cam ist not in the WLAN.
What does this Tell me?

FanDjango, 2025-03-30:
> I have an empty hack and ethernet file. But the cam ist not in the WLAN.
What does this Tell me?

It tells you the `hack.sh` actually ran.

Before using `hack.sh`, you should:

1. Get the camera setup and working in your WiFi using the app, using your WiFi SSID and password.
2. Check with a port scanner what IP address it has been given by your local DHCP server. Even better: Make your DHCP server assign a fixed IP address based on the cameras MAC adddress that the port scanner also shows.
3. After powering the camera off and on again, it should come up again and again and be able to be found at its IP address (try ping, for example).
4. Use the method described at the beginning of #34 to set up a telnet access (using `_ht_av_tuning.conf` and 'shadow'). Confirm that this works.
5. As the next step, now setup the SD card for using the "hack.sh" method. The camera should come up into WiFi at the known IP address.

If it does not, we could put more commands into `hack.sh` to make it report the result of an `ifconfig` command and store that on the SD card so that we can see it.

Fmaulick, 2025-03-31:
how did you make the shadow file?
You say its encrypded.

FanDjango, 2025-03-31:
I did not make it (although one can "make" such a file easily).

Create a user on a linux system, set his password to some string and then copy the "shadow" from there, is one way to do it.

The "shadow" file thus contains the encrypted password data for user "root". It is as if you had created a user "root" and set his password. It is done this way so that passwords for users are **not** stored in clear text but can still be checked on log-in etc.

piotrpot, 2025-03-31:
I have a similar issue with the same device and firmware version. I've logged the output of ifconfig before and after running your fix @FanDjango .

```
# Reset wifi to client mode
killall udhcpd
ifconfig wlan0 0.0.0.0
/usr/sbin/ethernet.sh
```

Before:
```
lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

wlan0     Link encap:Ethernet  HWaddr 78:22:88:1E:08:D8  
          inet addr:172.14.10.1  Bcast:172.14.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
```

After:
```
lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

```

FanDjango, 2025-03-31:
Well done!

Can you check:

Before and after this command?
```
/usr/sbin/ethernet.sh
```

Can you copy `/usr/sbin/ethernet.sh` to the sdcard and upload it here?

See what happens when you delete
```
killall udhcpd
ifconfig wlan0 0.0.0.0
```
(each one or even both).

This is the region where you need to debug...

My guess is that `ethernet.sh` is failing. Perhaps it does not exist? Check `/usr/sbin/service.sh` and look how **they** call it? Maybe it is renamed?

FanDjango, 2025-03-31:
Note: `hack.sh` relies on the Access Point config for startup and means that the WiFi comes up as an access point. In `hack.sh`, the WiFi access point is then "killed", the DHCP server is killed, and then the WiFi is "re-initialised` this time as a client.

For this to work, it relies on this being successully removed:

```
# Remove AP mode file
rm -f /tmp/_ht_ap_mode.conf
```

which makes all the subsequent logic understand that the Wifi and other software now must act like a configured client.

FanDjango, 2025-03-31:
Put this somewhere earlier for testing:
```
(while true; do /tmp/sd/custom.sh; sleep 10; done ) < /dev/null >& /dev/null &
```
It will start the telnet server on port 24 as described in #34. That way you might play around with the partly initialised system.

FanDjango, 2025-03-31:
You need to check the success and plausibility of each and every line in the later parts of `hack.sh` using the telnet access (with the _ht_av_tuning.conf and shadow method and/or by writing debug info to the sdcard).

Concentrating first on finding out why WiFi doesn't come back up. Then the later stuff.

Disregard my previous idea with starting telnet, **LOL you haven't got WiFi** so you couldn't use it anyway.

Fmaulick, 2025-03-31:
> I'm on firmware version **V3.2863.94**
> 
> After following the guide and using the files provided here, I cannot get any rtsp stream nor web interface.
> 
> **What I did:**
> 
> I used a SD card I formated through the app. I downloaded the repository files, extract the zip and copied the contents of the mmc directory into the SD card. I downloaded **busybox** from this link to the SD card: https://github.com/guino/LSC1080P/blob/main/mmc/busybox?raw=true I downloaded **motor.zip** (https://github.com/user-attachments/files/17739351/motor.zip) and replaced **motor.cgi** in the cgi-bin folder on the SD card.
> 
> I did not touched **http.conf**, **log_parser.sh** or **cleanup.cgi** on the SD card as I want to test first. I inserted the SD card on the device while powered off and let it boot up normally, I waited until I could see the device online on the phone app.
> 
> Then I powered off the device, took the SD card, downloaded **anyka_ipc_v3_patched_test.zip** (https://github.com/user-attachments/files/17738739/anyka_ipc_v3_patched_test.zip), extracted it, renamed **anyka_ipc_v3_patched_test_3** to **anyka_ipc_rtsp**, put **anyka_ipc_rtsp** at the root of the SD card.
> 
> Don't know if necessary, I also downloaded the 2.10.36 update file from tuya servers(https://fireware.tuyaeu.com:1443/smart/firmware/upgrade/ay1541668973821t35pE/165966791961ed11009a7.bin), extracted it and copied **libavssdkbeta.so** to the root of the SD card. I ejected the card correctly inserted the SD card on the device while powered off and let it boot up normally.
> 
> The camera works with the app, I cannot get any rtsp stream nor access the web interface.
> 
> I tried : `rtsp://user:password@192.168.178.xx/main_ch`, `rtsp://192.168.178.xx/main_ch` with username and password filled in Happytime rtsp client v6.1, I get "Connect failed".
> 
> I also cannot access : `http://user:password@192.168.178.xx:8080/index.html`, `http://user:password@192.168.178.xx:8080/ptz.html` or `http://192.168.178.xx:8080/ptz.html`, `http://192.168.178.xx:8080/ptz.html`.
> 
> **Nmap scan says:**
> 
> Scan (normal) Results of: 192.168.178.xx Starting Nmap 7.70 ( https://nmap.org ) at 2024-11-25 18:13 CET Nmap scan report for 192.168.178.xx Host is up (0.041s latency). Not shown: 998 closed ports PORT STATE SERVICE **23/tcp open telnet** **6668/tcp open irc**
> 
> Nmap done: 1 IP address (1 host up) scanned in 4.01 seconds
> 
> **Detailed scan says:**
> 
> Scan (detail) Results of: 192.168.178.xx Starting Nmap 7.70 ( https://nmap.org ) at 2024-11-25 17:57 CET Nmap scan report for 192.168.178.xx Host is up (0.051s latency). Not shown: 998 closed ports PORT STATE SERVICE VERSION **23/tcp open telnet BusyBox telnetd** **6668/tcp open irc?** | fingerprint-strings: | DNSVersionBindReqTCP, Help, Socks5: | \x0b | GenericLines, NULL: | \x0b | ]v\xc4! | GetRequest: | \x0b | u/9~!F | RTSPRequest: | \x0b |_ ]vNJq 1 service unrecognized despite returning data. If you know the service/version, please submit the following fingerprint at https://nmap.org/cgi-bin/submit.cgi?new-service : SF-Port6668-TCP:V=7.70%I=7%D=11/25%Time=6744AC74%P=arm-unknown-linux-gnuea SF:bihf%r(NULL,6B,"\0\0U\xaa\0\0\0\0\0\0\0\x08\0\0\0[\0\0\0\x003.3\0\0\0 SF:\0\0\0\xa8\x83\0\0\0\x01\xdfr~\xe9z\xd2Ny\xdf\x8a\0\xae:\xdb\xac@\xfc5 SF:+5\x0f\xeaN\xe7\xd3\xe5\x07\xec\xa4z\xfb\xe9N.\xc0x\xc6\xa4\xffj~z\xf0 SF:b\x8f\xfd\x9a\xa5\xb6\xd5\xac\xde\xa2B\xb9\x17\x82\xe4\xa6\xd4|\xfew\x SF:82\x9b\xa7)J\0\0\xaaU")%r(GenericLines,6B,"\0\0U\xaa\0\0\0\0\0\0\0\x08 SF:\0\0\0[\0\0\0\x003.3\0\0\0\0\0\0\xa8\x83\0\0\0\x01\xdfr~\xe9z\xd2Ny\x SF:df\x8a\0\xae:\xdb\xac@\xfc5+5\x0f\xeaN\xe7\xd3\xe5\x07\xec\xa4z\xfb\xe SF:9N.\xc0x\xc6\xa4\xffj~z\xf0b\x8f\xfd\x9a\xa5\xb6\xd5\xac\xde\xa2B\xb9 SF:x17\x82\xe4\xa6\xd4|\xfew\x82\x9b\xa7)J\0\0\xaaU")%r(HTTPOptions,5B," SF:\0\0U\xaa\0\0\0\0\0\0\0\x08\0\0\0K\0\0\0\x003.3\0\0\0\0\0\0\xa8\x87\0 SF:0\0\x01\x03\xbbA\x17\xd2|]\x88\xfe\x96\xa3[\x89X\xea>\xef\x1c\x1f\xa SF:8\xa3\xe2rc\x94\xcc\xc4\x9b\x06\x7f\xa9\x8eX\xc5G\xc1J\xcd?S\xcdF'2\n SF:xe7\x82K\xdcE\xe6\x87\0\0\xaaU")%r(RPCCheck,6B,"\0\0U\xaa\0\0\0\0\0\0\0 SF:\x08\0\0\0[\0\0\0\x003.3\0\0\0\0\0\0\xa8\x88\0\0\0\x01\xdfr~\xe9z\xd2 SF:Ny\xdf\x8a\0\xae:\xdb\xac@\xfc5+5\x0f\xeaN\xe7\xd3\xe5\x07\xec\xa4z\xf SF:b\xe9N.\xc0x\xc6\xa4\xffj~z\xf0b\x8f\xfd\x9a\xa5\0s\xc1\x1di\x8c\xf5\x SF:ee[\xf0\xb4N\x8c\x08Z\xdc\x08\x064\x9b\0\0\xaaU")%r(DNSStatusRequestTC SF:P,6B,"\0\0U\xaa\0\0\0\0\0\0\0\x08\0\0\0[\0\0\0\x003.3\0\0\0\0\0\0\xa8 SF:\x89\0\0\0\x01\xdfr~\xe9z\xd2Ny\xdf\x8a\0\xae:\xdb\xac@\xfc5+5\x0f\xea SF:N\xe7\xd3\xe5\x07\xec\xa4z\xfb\xe9N.\xc0x\xc6\xa4\xffj~z\xf0b\x8f\xfd SF:x9a\xa5\xe9\x96[\xd2\x02)\xba\xcb+\xbe\x83f=8\xc4Gp\xb6\xed\x9b\0\0 SF:xaaU")%r(TLSSessionReq,6B,"\0\0U\xaa\0\0\0\0\0\0\0\x08\0\0\0[\0\0\0\x0 SF:03.3\0\0\0\0\0\0\xa8\x8a\0\0\0\x01\xdfr~\xe9z\xd2Ny\xdf\x8a\0\xae:\xdb SF:\xac@\xfc5+5\x0f\xeaN\xe7\xd3\xe5\x07\xec\xa4z\xfb\xe9N.\xc0x\xc6\xa4 SF:\xffj~z\xf0b\x8f\xfd\x9a\xa52O\xcaT\xba#\xaa\x93E\xdf\t\x99\xc8km\xebH SF:xfb$J\0\0\xaaU")%r(SMBProgNeg,6B,"\0\0U\xaa\0\0\0\0\0\0\0\x08\0\0\0[ SF:0\0\0\x003.3\0\0\0\0\0\0\xa8\x8b\0\0\0\x01\xdfr~\xe9z\xd2Ny\xdf\x8a\0 SF:xae:\xdb\xac@\xfc5+5\x0f\xeaN\xe7\xd3\xe5\x07\xec\xa4z\xfb\xe9N.\xc0x SF:\xc6\xa4\xffj~z\xf0b\x8f\xfd\x9a\xa523\x0c\xdc\xeb{\xa0\x90Z\xd8P\xe8\x SF:a0\x9a\xcc\x1dsc\x99l\0\0\xaaU"); Service Info: Host: anyka
> 
> So it seems only 2 ports are opened.
> 
> Any idea what I did wrong? Can someone help?

that part of #27  helped. and now it is working.
thank you @FanDjango for helping me It took you to make me understand.

closed 2025-03-31 by Fmaulick

FanDjango, 2025-03-31:
@piotrpot Open a new issue if you want