guino/LSCOutdoor1080P #36: Rotatable Camera V3.2863.105

Joshicopter, 2025-04-20:
Hi,

I have a rotatable Camera with V3.2863.105.
I tried the steps:
Option 1 -> 1-6
Option 3
Option 1 -> 7- 16

The only thing I recognize is the first Bootsound and after that I hear only some silent sounds throw the speaker.
No Wifi or anything else.
I also tried a different Sd Card.

I also tried #27 but then there is no Bootsound and the size of anyka_ipc and anyka_ipc_rtsp is different.

Can someone help me?

buurned, 2025-04-20:
Did you find this Link? 
[https://community.gladysassistant.com/t/tuto-integrer-une-camera-lsc-dans-gladys-via-rtsp/9250/11](https://community.gladysassistant.com/t/tuto-integrer-une-camera-lsc-dans-gladys-via-rtsp/9250/11)
Yesterday I used this tutorial to make my wifi dual band rotatable camera rtsp available.

It had v.3.2863.94 now it is on v.3.2863.93.

Joshicopter, 2025-04-20:
Thank u for the link. The described way does not work. There is another User in this Post with the same Version as mine. 
There is a possible change in the cameras older than March 2025.

dynasource, 2025-04-24:
Can confirm it al works for both the LSC 360 outdoor (v3.2162.99) and 360 indoor!
- telnet works with the ankya login (root/telnet)
- step 3 succeeds
- vlc video playing at rtsp://ip/main_ch

It looks like the original firmware is ignored during the miniSD boot

rosenrot00, 2025-04-27:
> Can confirm it al works for both the LSC 360 outdoor (v3.2162.99) and 360 indoor!
> 
> * telnet works with the ankya login (root/telnet)
> * step 3 succeeds
> * vlc video playing at rtsp://ip/main_ch
> 
> It looks like the original firmware is ignored during the miniSD boot

Which method worked for you? The one described here in option 1 or the one from this link (https://community.gladysassistant.com/t/tuto-integrer-une-camera-lsc-dans-gladys-via-rtsp/9250/11) described above?

Thanks in advance.

rosenrot00, 2025-05-04:
> Thank u for the link. The described way does not work. There is another User in this Post with the same Version as mine, describing the Same Problem. There is a possible change in the cameras older than March 2025.

Did you manage to get it working? I just got my cameras up and running with RTSP

rosenrot00, 2025-05-22:
> Hello,
> 
> I have RTSP and motors working using this patched anyka_ipc_rtsp [#27 (comment)](https://github.com/guino/LSCOutdoor1080P/issues/27#issuecomment-2474702173)
> 
> But the app break and the blue led is blinking every seconds.
> 
> Since it's just a camera for my 3D printer, I don't really need other features than RTSP through home assistant but maybe someone know how I can setup the night vision ON forever
> 
> /usr/share # cat VERSION 
> 3.2863.105.28
> /usr/share # cat BUNDLE 
> HT_IPC286KSD3M_TUYA_AK3918EV330

Could you share your original firmware? It is stored on your sd card and it is called anyka_ipc

DataPocalypse42, 2025-05-22:
> 
> Could you share your original firmware? It is stored on your sd card and it is called anyka_ipc
>

Yes, here it is!

[anyka_ipc_original_3.2863.105.28.zip](https://github.com/user-attachments/files/20389383/anyka_ipc_original_3.2863.105.28.zip)

DataPocalypse42, 2025-05-23:
I'm also wondering if it's possible to bypass the motors initialization. 
When I tried to read serial datas, the IPC enter in a bootloop, which seems the motors can detect if the camera is disassembled.

Ideally I would like to make a lighter housing to fix the camera along the printer Z axis without motors.

rosenrot00, 2025-05-25:
Could you try if this patched version gives you an rtsp stream and the app still works? just replace it with the existing anyka_ipc_rtsp file on the emmc.

[anyka_ipc_patched_3.2863.105.28.zip](https://github.com/user-attachments/files/20429873/anyka_ipc_patched_3.2863.105.28.zip)

DataPocalypse42, 2025-05-26:
> > I'm also wondering if it's possible to bypass the motors initialization. When I tried to read serial datas, the IPC enter in a bootloop, which seems the motors can detect if the camera is disassembled.
> > Ideally I would like to make a lighter housing to fix the camera along the printer Z axis without motors.
> 
> If you want to turn off PTZ, add this command before mounting _ht_hw_settings.ini in your hack.sh file.
> 
> `sed -i 's/^ptz_support *= *.*/ptz_support = 0/' /tmp/_ht_hw_settings.ini`

Thank you, I got strange bootloop with the sed command, even after removing it from hack.sh.

Finally I got better result by copying _ht_hw_settings.ini and _ht_sw_settings.ini to and mount from SD card directly.

So disabling PTZ no more bootloop with motors unplugged.

For the blinking blue LED, I put this at end of custom.sh ( this do not work in hack.sh, should be in the loop )
```bash
## Disable blue LED
echo 0 > /sys/class/leds/state_led/brightness
```

And for the night vision ON by default, it's this setting in _ht_sw_settings.ini :
It is 0 by default, 1 do nothing, tried 2 and it worked. 
```ini
enum_night_mode                = 2
```

Another interesting setting in _ht_hw_settings.ini for those who have issues with H265, you can chose another encoder
```ini
;[0,2]:0 h264, 1 mjpeg, 2 h265
sub_enc_out_type = 0
```

Let me know if you need to try patched IPC, but with PTZ disabled

butsify, 2025-07-31:
Hi @rosenrot00!

I did try your v2 and it works for me. I got working rtsp with audio and also the app does work. What i can't get to work is ptz.html and MQTT.