guino/LSCOutdoor1080P #38: Patched v2.10.36 and v3.303.70

rosenrot00, 2025-05-17:
If anyone is interested, I patched these two versions without breaking the tuya apps.

As here is little to none activity I'm not just posting the files here but want to know if there is a real request for it.

Dbrunello, 2025-08-24:
I'm interested

ronaldgit, 2026-02-17:
Yes, I'm interested too.

edennede, 2026-10-23:
Hello @rosenrot00,

I'm interested in the v3.202.70 patch too. I have an LSC Smart Connect outdoor camera (mains-powered, bought at Action) running exactly this firmware, and I'd like to use RTSP with Frigate/Home Assistant while keeping the Tuya app working.

Could you share the files and the instructions (including the MD5 of the original anyka_ipc file the patch applies to)? I'm happy to report back on the result.

Thanks in advance!

edennede, 2026-10-25:
Hi, could someone please share the root files that worked for your LSC Outdoor 1080p camera? Thanks!

edennede, 2026-10-27:
Hi, someone please ? I dont' want to buy a new camera  😔

edennede, 2026-11-03:
Someone ?