guino/LSCOutdoor1080P #39: Rotatable camera indoor V3.2863.105

lapuff, 2025-05-17:
Hello,

I'm trying to install the program without success.
I followed option 1, but I don't have access via TELNET or RTSP.
However, if I install only the two files required for TELNET, it works fine.

Has anyone managed to get it working with version V3.28.63.105?
Thanks!

rosenrot00, 2025-05-18:
Can you share your original firmware version that was copied to the SD card?

lapuff, 2025-05-19:
How do we do it? When I copy the mmc folder and the busybox file to the root of the SD card, it seems like nothing happens on the camera.

Can you explain how to retrieve the firmware?

rosenrot00, 2025-05-19:
The anyka_ipc file should get copied on your SD card. So when you follow instructions and copy the files to the root of the SD card; insert it; boot the camera until fully started; insert the SD into your computer, this file should show up.

rosenrot00, 2025-05-19:
That is weird as here it seemed to work. https://github.com/guino/LSCOutdoor1080P/issues/36

To clarify, the mmc folder's content and busybox is in the root directoy? Not /mmc...?

lapuff, 2025-05-20:
Thank you very much for your help! The issue was that I was copying the mmc folder itself instead of its contents… Now everything is working correctly. I really appreciate it!

One last question: how can I change the username and password? Is it done via Telnet or by editing the configuration files?

lapuff, 2025-05-21:
I just found the answer in the Telnet notes, thanks!